ANYbotics/elevation_mapping #29: Any path planning ROS package compatible with this grid map?

robotfan4, 2016-07-07:
This work is excellent for legged robots, however I can't find any path planning ROS package working with the grid map, Any suggestion will be appreciated. I known the grid map can be converted to pointcloud2 or occupied map.

pfankhauser, 2016-07-07:
Thanks for you interest @robotfan4! This software is still very young and I'm not aware of any open-source navigation planning packages that build up directly on the grid map types (our internal planner is not (yet) publicly available). However, as you pointed out, you can easily transform the maps to other more common types such as occupancy grids. For this kind of work, you might consider also using our [traversability estimation package](https://github.com/ethz-asl/traversability_estimation) which is based on grid maps.

closed 2016-07-07 by pfankhauser

zhangk22, 2019-04-08:
Hello @pfankhauser ,

I am also interested in this topic. Since your last answer is in 2016, I wonder whether there is any update. Additionally, as you mentioned that the traversability estimation package can produce occupancy grids. Do you know any good motion planning package that works with it?

Thanks

pfankhauser, 2019-04-08:
There's a [PR](https://github.com/ANYbotics/elevation_mapping/pull/84) with a plugin to convert elevation maps to `costmap_2d` that will make it compatible with the navigation stack. Maybe this helps?

jorgemia, 2020-11-10:
Have there been any further updates or recommendations? I'm trying to use a 4WD robot for outdoor navigation but not sure if I can use this package with something else for 3D navigation or I should just stick to the normal navigation stack.

maximilianwulf, 2020-11-11:
ROS2 will integrate grid_maps natively into its navigation stack. But this is still ongoing work. 

I guess the mentioned recommendations are still valid options. 

3D navigation will be unrealistic with a 2.5D map though.