ANYbotics/elevation_mapping #30: Udpated master to latest Kindr version

pfankhauser, 2016-07-18:
As of now, the master branch is _not_ compatible with the [Kindr](https://github.com/ethz-asl/kindr) master. We are working on a bigger update, so in the meanwhile please use the [feature/obstacle_detection](https://github.com/ethz-asl/elevation_mapping/tree/feature/obstacle_detection) branch for compatibility.

Nurgak, 2016-09-10:
Any news on an update? Even with the `feature/obstacle_detection` branch I cannot get `elevation_mapping` compiled because of a kindr issue. I'm using catkin tools:

```
Errors     << elevation_mapping:cmake /home/karl/catkin_ws/logs/elevation_mapping/build.cmake.001.log                                                         
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "kindr_ros" with
  any of the following names:

    kindr_rosConfig.cmake
    kindr_ros-config.cmake

  Add the installation prefix of "kindr_ros" to CMAKE_PREFIX_PATH or set
  "kindr_ros_DIR" to a directory containing one of the above files.  If
  "kindr_ros" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:15 (find_package)

CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "kindr_ros" with
  any of the following names:

    kindr_rosConfig.cmake
    kindr_ros-config.cmake

  Add the installation prefix of "kindr_ros" to CMAKE_PREFIX_PATH or set
  "kindr_ros_DIR" to a directory containing one of the above files.  If
  "kindr_ros" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:15 (find_package)

cd /home/karl/catkin_ws/build/elevation_mapping; catkin build --get-env elevation_mapping | catkin env -si  /usr/bin/cmake /home/karl/catkin_ws/src/elevation_mapping/elevation_mapping --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/karl/catkin_ws/devel/.private/elevation_mapping -DCMAKE_INSTALL_PREFIX=/home/karl/catkin_ws/install; cd -
```
Can you perhaps update the installation instructions? I can't seem to get anything working with the one provided.

pfankhauser, 2016-09-12:
Dear @Nurgak,
I have updated the master branch with the newest algorithmic and compatibility changes and adapted the README accordingly. Could you please check if things work for you now?
Thanks, Peter

Nurgak, 2016-09-12:
Hi @pfankhauser, thanks for the quick response, however I am still unable to build the package, I get the following warnings and error:

```
_____________________________________________________________________________________________________________________________________
Warnings   << kindr_rviz_plugins:make /home/karl/catkin_ws/logs/kindr_rviz_plugins/build.make.000.log                                
/home/karl/catkin_ws/src/kindr_ros/kindr_rviz_plugins/include/kindr_rviz_plugins/VectorAtPositionVisual.hpp:0: Note: No relevant classes found. No output generated.
cd /home/karl/catkin_ws/build/kindr_rviz_plugins; catkin build --get-env kindr_rviz_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.....................................................................................................................................
Finished  <<< kindr_rviz_plugins                                     [ 26.3 seconds ]                                                
_____________________________________________________________________________________________________________________________________
Warnings   << multi_dof_joint_trajectory_rviz_plugins:make /home/karl/catkin_ws/logs/multi_dof_joint_trajectory_rviz_plugins/build.make.000.log
/home/karl/catkin_ws/src/kindr_ros/multi_dof_joint_trajectory_rviz_plugins/include/multi_dof_joint_trajectory_rviz_plugins/MultiDOFJointTrajectoryPointVisual.hpp:0: Note: No relevant classes found. No output generated.
/home/karl/catkin_ws/src/kindr_ros/multi_dof_joint_trajectory_rviz_plugins/include/multi_dof_joint_trajectory_rviz_plugins/MultiDOFJointTrajectoryPointConnectionVisual.hpp:0: Note: No relevant classes found. No output generated.
cd /home/karl/catkin_ws/build/multi_dof_joint_trajectory_rviz_plugins; catkin build --get-env multi_dof_joint_trajectory_rviz_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.....................................................................................................................................
Finished  <<< multi_dof_joint_trajectory_rviz_plugins                [ 38.4 seconds ]                                                
_____________________________________________________________________________________________________________________________________
Errors     << elevation_mapping:make /home/karl/catkin_ws/logs/elevation_mapping/build.make.000.log                                  
/home/karl/catkin_ws/src/elevation_mapping/elevation_mapping/src/ElevationMap.cpp: In member function ‘bool elevation_mapping::ElevationMap::fuse(const Index&, const Index&, bool)’:
/home/karl/catkin_ws/src/elevation_mapping/elevation_mapping/src/ElevationMap.cpp:285:65: error: ‘class grid_map::EllipseIterator’ has no member named ‘getSubmapSize’
     const unsigned int maxNumberOfCellsToFuse = ellipseIterator.getSubmapSize().prod();
                                                                 ^
make[2]: *** [CMakeFiles/elevation_mapping_library.dir/src/ElevationMap.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/elevation_mapping_library.dir/all] Error 2
make: *** [all] Error 2
cd /home/karl/catkin_ws/build/elevation_mapping; catkin build --get-env elevation_mapping | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

pfankhauser, 2016-09-12:
Could you please verify that you are using the latest version (1.4.0) of the Grid Map package from either [here](https://github.com/ethz-asl/grid_map) or the ROS PPA?

Nurgak, 2016-09-12:
I just tried, some warnings disappeared, another appeared, but the same error persists:

```
_____________________________________________________________________________________________________________________________________
Errors     << elevation_mapping:make /home/karl/catkin_ws/logs/elevation_mapping/build.make.001.log                                  
/home/karl/catkin_ws/src/elevation_mapping/elevation_mapping/src/ElevationMap.cpp: In member function ‘bool elevation_mapping::ElevationMap::fuse(const Index&, const Index&, bool)’:
/home/karl/catkin_ws/src/elevation_mapping/elevation_mapping/src/ElevationMap.cpp:285:65: error: ‘class grid_map::EllipseIterator’ has no member named ‘getSubmapSize’
     const unsigned int maxNumberOfCellsToFuse = ellipseIterator.getSubmapSize().prod();
                                                                 ^
make[2]: *** [CMakeFiles/elevation_mapping_library.dir/src/ElevationMap.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/elevation_mapping_library.dir/all] Error 2
make: *** [all] Error 2
cd /home/karl/catkin_ws/build/elevation_mapping; catkin build --get-env elevation_mapping | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.....................................................................................................................................
Failed     << elevation_mapping:make                                 [ Exited with code 2 ]                                          
Failed    <<< elevation_mapping                                      [ 34.5 seconds ]                                                
Abandoned <<< grid_map_demos                                         [ Unrelated job failed ]                                        
Abandoned <<< elevation_mapping_demos                                [ Unrelated job failed ]                                        
_____________________________________________________________________________________________________________________________________
Warnings   << grid_map_rviz_plugin:make /home/karl/catkin_ws/logs/grid_map_rviz_plugin/build.make.000.log                            
/home/karl/catkin_ws/src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/GridMapVisual.hpp:0: Note: No relevant classes found. No output generated.
/home/karl/catkin_ws/src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter.h:486: Note: No relevant classes found. No output generated.
cd /home/karl/catkin_ws/build/grid_map_rviz_plugin; catkin build --get-env grid_map_rviz_plugin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

It seems `getSubmapSize` is undefined...

pfankhauser, 2016-09-12:
Please excuse me, I overlooked that. This method has been added very recently and has not been pushed to the PPA yet. Could you please try with the master branch from Github?

Nurgak, 2016-09-12:
I'm afraid I don''t follow... my `elevation_mapping` master branch is up to date, so the output is still erroneous.

pfankhauser, 2016-09-12:
Sorry, I meant the master branch of Grid Map at https://github.com/ethz-asl/grid_map

Nurgak, 2016-09-12:
Just to be sure I checked them all, everything is up to date: `elevation_mapping`, `grid_map` and `kindr_ros`... quite strange.

pfankhauser, 2016-09-13:
Could you check the following file: `grid_map_core/include/grid_map_core/iterators/EllipseIterator.hpp`, Line 74 should have the declaration

    const Size& getSubmapSize() const;

as [here](https://github.com/ethz-asl/grid_map/blob/master/grid_map_core/include/grid_map_core/iterators/EllipseIterator.hpp#L74).

Nurgak, 2016-09-13:
I got it, I needed to remove `/devel` and `/build` and start over again. However this resulted in 2 warnings (they look benign though) and I cannot execute the simple demo example, the `elevation_mapping` node keeps crashing and there are no log files created to debug.

The error when executing `roslaunch elevation_mapping_demos simple_demo.launch`:
```
[elevation_mapping-1] process has died [pid 11623, exit code -11, cmd /home/karl/catkin_ws/devel/lib/elevation_mapping/elevation_mapping __name:=elevation_mapping __log:=/home/karl/.ros/log/71ebb01e-7981-11e6-a952-ecf4bb6e33e2/elevation_mapping-1.log].
log file: /home/karl/.ros/log/71ebb01e-7981-11e6-a952-ecf4bb6e33e2/elevation_mapping-1*.log
```

The warnings during compilation:
```
____________________________________________________________________________________________________________________________________________________________________________
Warnings   << multi_dof_joint_trajectory_rviz_plugins:make /home/karl/catkin_ws/logs/multi_dof_joint_trajectory_rviz_plugins/build.make.006.log                             
/home/karl/catkin_ws/src/kindr_ros/multi_dof_joint_trajectory_rviz_plugins/include/multi_dof_joint_trajectory_rviz_plugins/MultiDOFJointTrajectoryPointVisual.hpp:0: Note: No relevant classes found. No output generated.
/home/karl/catkin_ws/src/kindr_ros/multi_dof_joint_trajectory_rviz_plugins/include/multi_dof_joint_trajectory_rviz_plugins/MultiDOFJointTrajectoryPointConnectionVisual.hpp:0: Note: No relevant classes found. No output generated.
cd /home/karl/catkin_ws/build/multi_dof_joint_trajectory_rviz_plugins; catkin build --get-env multi_dof_joint_trajectory_rviz_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
............................................................................................................................................................................
____________________________________________________________________________________________________________________________________________________________________________
Warnings   << kindr_rviz_plugins:make /home/karl/catkin_ws/logs/kindr_rviz_plugins/build.make.006.log                                                                       
/home/karl/catkin_ws/src/kindr_ros/kindr_rviz_plugins/include/kindr_rviz_plugins/VectorAtPositionVisual.hpp:0: Note: No relevant classes found. No output generated.
cd /home/karl/catkin_ws/build/kindr_rviz_plugins; catkin build --get-env kindr_rviz_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
............................................................................................................................................................................
____________________________________________________________________________________________________________________________________________________________________________
Warnings   << grid_map_rviz_plugin:make /home/karl/catkin_ws/logs/grid_map_rviz_plugin/build.make.005.log                                                                   
/home/karl/catkin_ws/src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/GridMapVisual.hpp:0: Note: No relevant classes found. No output generated.
/home/karl/catkin_ws/src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter.h:486: Note: No relevant classes found. No output generated.
cd /home/karl/catkin_ws/build/grid_map_rviz_plugin; catkin build --get-env grid_map_rviz_plugin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
............................................................................................................................................................................
```

pfankhauser, 2016-09-20:
Has this resolved with https://github.com/ethz-asl/elevation_mapping/issues/33?

Nurgak, 2016-09-20:
Yes, the process does not die anymore #33, solved the issue.

closed 2016-12-05 by pfankhauser