ANYbotics/elevation_mapping #32: Wrong paths for the ground truth demo

Nurgak, 2016-09-13:
I noticed the paths for the ground truth demo are wrong and the launch file does not work.

This is what is currently written:
```
<launch>
 	<node pkg="elevation_mapping" type="elevation_mapping" name="elevation_mapping" output="screen">
    	 <rosparam command="load" file="$(find elevation_mapping)/parameters/robots/ground_truth.yaml" />
    	 <rosparam command="load" file="$(find elevation_mapping)/parameters/elevation_maps/long_range.yaml" />
    	 <rosparam command="load" file="$(find elevation_mapping)/parameters/sensor_processors/perfect.yaml" />
    </node>
</launch>
```

This is what is should be:
```
<launch>
 	<node pkg="elevation_mapping" type="elevation_mapping" name="elevation_mapping" output="screen">
    	 <rosparam command="load" file="$(find elevation_mapping_demos)/config/robots/ground_truth_demo.yaml" />
    	 <rosparam command="load" file="$(find elevation_mapping_demos)/config/elevation_maps/long_range.yaml" />
    	 <rosparam command="load" file="$(find elevation_mapping)/config/sensor_processors/perfect.yaml" />
    </node>
</launch>
```

wangjio, 2016-09-13:
when I try to run node "elevation_mapping", i was always told that "process has died", have you ever get this problem?

wangjio, 2016-09-13:
I saw your problem in the last issue. I have the same problem with you. Have you solved that problem?

Nurgak, 2016-09-13:
Not yet, I can't get any specific error output so I don't know where the issue is.

When looking at the roslaunch log file the process shows that it dies with an error code -11:

```
[roslaunch][ERROR] 2016-09-13 16:00:20,924: [elevation_mapping-2] process has died [pid 31736, exit code -11, cmd /home/karl/catkin_ws/devel/lib/elevation_mapping/elevation_mapping __name:=elevation_mapping __log:=/home/karl/.ros/log/71ebb01e-7981-11e6-a952-ecf4bb6e33e2/elevation_mapping-2.log].
log file: /home/karl/.ros/log/71ebb01e-7981-11e6-a952-ecf4bb6e33e2/elevation_mapping-2*.log
[roslaunch.pmon][INFO] 2016-09-13 16:00:20,924: ProcessMonitor.unregister[elevation_mapping-2] starting
[roslaunch.pmon][INFO] 2016-09-13 16:00:20,924: ProcessMonitor.unregister[elevation_mapping-2] complete
[roslaunch.pmon][INFO] 2016-09-13 16:00:23,228: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, started daemon 140012150454016)>
[roslaunch.pmon][INFO] 2016-09-13 16:00:23,328: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 140012150454016)>
[roslaunch.pmon][INFO] 2016-09-13 16:00:23,329: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 140012150454016)>: remaining procs are [<roslaunch.nodeprocess.LocalProcess object at 0x7f571ee1ef10>, <roslaunch.nodeprocess.LocalProcess object at 0x7f571ee1e5d0>]
[roslaunch.pmon][INFO] 2016-09-13 16:00:23,329: ProcessMonitor exit: killing rviz-3
[roslaunch][INFO] 2016-09-13 16:00:23,329: [rviz-3] killing on exit
[roslaunch.pmon][INFO] 2016-09-13 16:00:23,329: ProcessMonitor exit: killing robot_state_publisher-1
[roslaunch][INFO] 2016-09-13 16:00:23,330: [robot_state_publisher-1] killing on exit
[roslaunch][INFO] 2016-09-13 16:00:23,330: process[rviz-3]: killing os process with pid[31737] pgid[31737]
[roslaunch][INFO] 2016-09-13 16:00:23,330: process[robot_state_publisher-1]: killing os process with pid[31735] pgid[31735]
[roslaunch][INFO] 2016-09-13 16:00:23,331: [robot_state_publisher-1] sending SIGINT to pgid [31735]
[roslaunch][INFO] 2016-09-13 16:00:23,330: [rviz-3] sending SIGINT to pgid [31737]
[roslaunch][INFO] 2016-09-13 16:00:23,331: [robot_state_publisher-1] sent SIGINT to pgid [31735]
[roslaunch][INFO] 2016-09-13 16:00:23,331: [rviz-3] sent SIGINT to pgid [31737]
[roslaunch][INFO] 2016-09-13 16:00:23,532: process[robot_state_publisher-1]: SIGINT killed with return value 0
[roslaunch][INFO] 2016-09-13 16:00:23,732: process[rviz-3]: SIGINT killed with return value 0
```

It says to check the `elevation_mapping` log files, but they are not created.

When I try `rosrun elevation_mapping elevation_mapping` then I get a `Segmentation fault (core dumped)`, although I don't know if I'm actually allowed to call it this way...

wangjio, 2016-09-13:
I have read 'elevation_mapping_node.cpp', and mabye there is something wrong with line 17, the constructor of class ElevationMapping. Because after i comment that line, Segmentation fault disappeared.

pfankhauser, 2016-12-05:
Paths have been corrected in https://github.com/ethz-asl/elevation_mapping/pull/36. Is this still an issue?

wojiaojiao, 2017-01-09:
please help me ,all the install step i was success ,but till "catkin_make run_tests_elevation_map_msg run_tests_elevation_mapping",it can't work.i "catkin_make"success!why?

wojiaojiao, 2017-03-07:
I have kinect and binocular camera, can tell me how to use these two devices running it, please, I need your detailed instructions @hitzg @Nurgak @dljsjr

TanyuSir, 2018-10-15:
> please help me ,all the install step i was success ,but till "catkin_make run_tests_elevation_map_msg run_tests_elevation_mapping",it can't work.i "catkin_make"success!why?

I have encountered the same problem. Can you find a solution to the problem?