ANYbotics/elevation_mapping #33: elevation_mapping process has died

baishibona, 2016-09-16:
This is very useful however I was getting stuck on the elevation_mapping node, when I run:
rosrun elevation_mapping elevation_mapping
the process simple died immediately... saying: core dumped...
Even I define each topic name using "simple_demo.launch", it still does not run.
Is it because I was missing part of the topic? I did provide pointcloud2 pose and tf though.

Any suggestion on what should I try next?
Thanks!

Nurgak, 2016-09-20:
I found a solution: you need to build the package with `-DCMAKE_BUILD_TYPE=Release` argument, so for me it was `catkin build elevation_mapping -DCMAKE_BUILD_TYPE=Release`.

The node will stop throwing the segmentation fault error and work properly. I found it by following the [node debug procedures](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) and ended up on [this issue page](http://stackoverflow.com/questions/26346690/debug-seg-fault-in-boostmathlanczos-with-libpcl-surface).

baishibona, 2016-09-20:
@Nurgak Thank you so much! I will give it a try today and let you guys know how it goes. Hopefully we can close this issue soon.

pfankhauser, 2016-09-20:
Dear @baishibona and @Nurgak, thanks for reporting and investigating this. This is a problem I cannot currently solve as it is related to a problem with PCL/Boost and Debug mode/C++11. I will re-open this issue to keep track of the situation.

skohlbr, 2016-11-13:
I also ran into this issue, which as Peter said appears to be related to https://github.com/PointCloudLibrary/pcl/issues/619. For reference, this is the backtrace I'm getting:

```
(gdb) bt
#0  0x00007ffff1f46ae0 in boost::math::lanczos::lanczos_initializer<boost::math::lanczos::lanczos17m64, long double>::init::init() ()
   from /usr/lib/libpcl_sample_consensus.so.1.7
#1  0x00007ffff1f1e5ce in ?? () from /usr/lib/libpcl_sample_consensus.so.1.7
#2  0x00007ffff7dea10a in call_init (l=<optimized out>, argc=argc@entry=3, 
    argv=argv@entry=0x7fffffff8348, env=env@entry=0x7fffffff8368)
    at dl-init.c:78
#3  0x00007ffff7dea1f3 in call_init (env=<optimized out>, 
    argv=<optimized out>, argc=<optimized out>, l=<optimized out>)
    at dl-init.c:36
#4  _dl_init (main_map=0x7ffff7ffe1c8, argc=3, argv=0x7fffffff8348, 
    env=0x7fffffff8368) at dl-init.c:126
#5  0x00007ffff7ddb30a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#6  0x0000000000000003 in ?? ()
#7  0x00007fffffff87bc in ?? ()
#8  0x00007fffffff8801 in ?? ()
#9  0x00007fffffff881b in ?? ()
#10 0x0000000000000000 in ?? ()
```

Would a temporary fix be adding https://github.com/ethz-asl/pcl_catkin to the workspace and thus compiling PCL from source?

skohlbr, 2016-12-01:
Any updates/ideas for this problem? It seems it's gotten worse (?) for us, with getting loops of `std::bad_alloc` segfaults even when compiled in Release mode on multiple (but not all) machines we tried.

pfankhauser, 2016-12-05:
@skohlbr Have you tried building pcl from source with C++11? According to [here](https://github.com/PointCloudLibrary/pcl/issues/619#issuecomment-186854511) this should solve the problem. It's strange that the problem has gotten worse for you, especially since the problem is related to the Debug mode. Are you sure that this is the same problem?

wojiaojiao, 2017-03-07:
I have kinect and binocular camera, can tell me how to use these two devices running it, please, I need your detailed instructions @pfankhauser @baishibona @dljsjr

baishibona, 2017-03-12:
@wojiaojiao I think there is "starleth_kinect_demo.launch" which you might try with kinect, however not sure what to do with stereo cameras(depending on the driver for that), you might want to adjust your topic name for point cloud msg.
good luck.

wojiaojiao, 2017-03-14:
@baishibona thank you ,but i only have kinect1,but no IMU or GPS and so on，Can I run out？

wojiaojiao, 2017-03-14:
@baishibona @Nurgak @skohlbr @dljsjr i use kinect1 roslaunch starleth_kinect_demo.launch.but in rviz have nothing,the terminal show:
[ERROR] [1489454583.211567394]: Could not get pose information from robot for time 1489454583.211551. Buffer empty?
[ERROR] [1489454583.211595878]: Updating process noise failed.

how can i solve it, Please help me