ANYbotics/elevation_mapping #34: Nan value for elevation map, also terminated due to a 'std::bad_alloc'

baishibona, 2016-09-20:
I was turning in "velodyne_points", "pose" and "tf" to the elevation_mapping node, however didn't get a valid elevation map. When I do 
rostopic echo "elevation_map"
It shows a lot of Nan values looks like the following:

 nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan, nan]
outer_start_index: 109
inner_start_index: 61
---

and the elevation_map node got the following error message:

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Any suggestions?
Thanks!

skohlbr, 2016-12-01:
It's very likely this is the same problem as described in #33.

wojiaojiao, 2017-03-07:
I have kinect and binocular camera, can tell me how to use these two devices running it, please, I need your detailed instructions @hitzg @ZacharyTaylor @baishibona @skohlbr @dljsjr