ANYbotics/elevation_mapping #35: nan Value for elevation map

sivasriram2112, 2016-11-06:
The elevation map node is subscribed to /depth/points from asus xtion, tf and pose with covariance stamped, but rostopic echo elevation_mapping/elevation_map returns NAN.

skohlbr, 2016-11-24:
All cells not seen by the sensor IIRC contain NANs, so this is not really a error. In any case, you will need to provide a lot more information about the issue, ideally via a [Minimal Working Example](https://en.wikipedia.org/wiki/Minimal_Working_Example).

wojiaojiao, 2017-01-09:
please help me ,all the install step i was success ,but till "catkin_make run_tests_elevation_map_msg run_tests_elevation_mapping",it can't work.i "catkin_make"success!why?

wojiaojiao, 2017-03-07:
I have kinect and binocular camera, can tell me how to use these two devices running it, please, I need your detailed instructions

wojiaojiao, 2017-03-07:
@sivasriram2112