ANYbotics/elevation_mapping #39: The /elevation_mapping/elevation_map topic has NAN value

sarahTohidi, 2017-04-11:
I have set every parameter of this package in the launch file in elevation_mapping_demos and also I have set parameters in config files in both elevation_mapping and elevation_mapping_demos.
I have also checked the topics which this node subscribes and every one of them has correct values also, the rate of publishing data has been checked. 
The only topic that I can visualise is the /map_region which has a correct value. 
I have noticed some part of the code has //TODO, I don't know what is the problem.
Is there any help to figure it out?
I'll appreciate any help

sarahTohidi, 2017-04-14:
I've solved the problem by resetting all the config files then, I figured out there was my mistake to set a wrong parameter in one of the config files.

Now I have another problem which is the elevation_mapping crashes by _Exception error_ of Eigen as following:

**elevation_mapping: /usr/include/eigen3/Eigen/src/Eigenvalues/EigenSolver.h:347: Eigen::EigenSolver<_MatrixType>::EigenvectorsType Eigen::EigenSolver<_MatrixType>::eigenvectors() const [with _MatrixType = Eigen::Matrix<double, 2, 2>; Eigen::EigenSolver<_MatrixType>::EigenvectorsType = Eigen::Matrix<std::complex<double>, 2, 2>; typename Eigen::NumTraits<typename Derived::Scalar>::Real = double]: Assertion `m_eigenvectorsOk && "The eigenvectors have not been computed together with the eigenvalues."' failed.
[elevation_mapping-1] process has died [pid 23174, exit code -6, cmd /home/delnia/new_way/devel/lib/elevation_mapping/elevation_mapping __name:=elevation_mapping __log:=/home/delnia/.ros/log/3c3c5626-20d4-11e7-a3d1-b88a60a09600/elevation_mapping-1.log].
log file: /home/delnia/.ros/log/3c3c5626-20d4-11e7-a3d1-b88a60a09600/elevation_mapping-1*.log**

This is happening when I launch the 

> roslaunch elevation_mapping_demos simple_demo.launch

Seems that this is the main problem is this: **"The eigenvectors have not been computed together with the eigenvalues."** 

I'll appreciate any help to solve this. :)

sarahTohidi, 2017-04-18:
I have found that this error comes from this line[ 276];

    const double ellipseRotation(atan2(solver.eigenvectors().col(maxEigenvalueIndex).real()(1), solver.eigenvectors().col(maxEigenvalueIndex).real()(0)));

which in the ElevationMap.cpp in the elevation_mapping.

I've tried with try&catch but doesn't work.

I'll appreciate any help to solve it.

sarahTohidi, 2017-04-18:
I don't exactly know why you are getting it a default 1,0 value but when I made it permanent the problem solve and the error disappears.

I hope you give me a response if you find some time.

sarahTohidi, 2017-04-18:
The elevation mapping is now working but when I rotate and move forward (most of the times) the elevation mapping crashes.
With this error:

> elevation_mapping: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:118: Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType Eigen::DenseCoeffsBase<Derived, 0>::operator()(Eigen::Index, Eigen::Index) const [with Derived = Eigen::Matrix<float, -1, -1>; Eigen::DenseCoeffsBase<Derived, 0>::CoeffReturnType = const float&; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.

nlimpert, 2017-08-28:
Hello,

how exactly did you manage to fix the error in line [277] (now it is 277) of file ElevationMap.cpp?
I can verify that the Backtrace in GDB points at this line but I'm stuck there.
Over here the "ellipseRotation" is sometimes working fine with 0.0 or with small values ranging from -1 to 1.

Which permanent value did you set up?

BrunoEduardoCSantos, 2018-10-31:
@SaraTohidi Did you solve it?

duyipai, 2018-11-06:
Hello,
Same here, I am also stuck at line 277. Is covarianceMatrix bing [inf, 0; 0, inf] causing the problem? How can I fix it? Many thanks!

SaraTohidi, 2018-11-26:
Hello,
Unfortunately, I do not remember very clearly but I gave the 1 and 0 as a permanent value because sometimes this default changed and led in a crash. However, the problem with crashes never gone.
I was working on a project (traversal_map on my GitHub) and because of that I needed an elevation map,  but this package had many problems I could not put my time to fix them. So, I started to use the OctoMap package instead and customize it which worked well for me.
I hope this finds you well.

duyipai, 2018-11-27:
Resolved through using newer version of eigen3. Hope it helps others.

StanleyYake, 2019-03-28:
> I've solved the problem by resetting all the config files then, I figured out there was my mistake to set a wrong parameter in one of the config files.
> 
> Now I have another problem which is the elevation_mapping crashes by _Exception error_ of Eigen as following:
> 
> __elevation_mapping: /usr/include/eigen3/Eigen/src/Eigenvalues/EigenSolver.h:347: Eigen::EigenSolver<_MatrixType>::EigenvectorsType Eigen::EigenSolver<_MatrixType>::eigenvectors() const [with _MatrixType = Eigen::Matrix<double, 2, 2>; Eigen::EigenSolver<_MatrixType>::EigenvectorsType = Eigen::Matrix<std::complex, 2, 2>; typename Eigen::NumTraits::Real = double]: Assertion `m_eigenvectorsOk && "The eigenvectors have not been computed together with the eigenvalues."' failed. [elevation_mapping-1] process has died [pid 23174, exit code -6, cmd /home/delnia/new_way/devel/lib/elevation_mapping/elevation_mapping __name:=elevation_mapping __log:=/home/delnia/.ros/log/3c3c5626-20d4-11e7-a3d1-b88a60a09600/elevation_mapping-1.log]. log file: /home/delnia/.ros/log/3c3c5626-20d4-11e7-a3d1-b88a60a09600/elevation_mapping-1_.log_*
> 
> This is happening when I launch the
> 
> > roslaunch elevation_mapping_demos simple_demo.launch
> 
> Seems that this is the main problem is this: **"The eigenvectors have not been computed together with the eigenvalues."**
> 
> I'll appreciate any help to solve this. :)

Hello Bruno Santos, I have the same problem with you, Only  /map_region which has a correct value .What's your correct parameter in the config files. I am very appreciate if you could do me a favour to upload your param for me to take a look.

bestavra, 2022-10-08:
Hi, I've experienced the same _Exception error_ of Eigen as follows:
_elevation_mapping: /usr/include/eigen3/Eigen/src/Eigenvalues/EigenSolver.h:348: Eigen::EigenSolver<_MatrixType>::EigenvectorsType Eigen::EigenSolver<_MatrixType>::eigenvectors() const [with _MatrixType = Eigen::Matrix<double, 2, 2>; Eigen::EigenSolver<_MatrixType>::EigenvectorsType = Eigen::Matrix<std::complex<double>, 2, 2>; typename Eigen::NumTraits<typename Derived::Scalar>::Real = double]: Assertion `m_eigenvectorsOk && "The eigenvectors have not been computed together with the eigenvalues."' failed._

Setup: Ubuntu 20.04, ROS-Noetic

It has to do something with the map initialization. Without map initialization, each of the following cases runs fine. 
The waffle demo example runs fine even with the map initialization though.
The problem appears when with nearly the same .yaml file as in the waffle demo (except using lidar pcl data) the Husky robot with LIO-SAM is used for pose estimation. If using Husky with the default localization (robot_localization pkg), the problem disappears.

Back to the troubling case when LIO-SAM is used. When running a custom launch file including elevation_mapping, everything is fine, even the map initialization (checking by adding ROS_INFO on line 647 in ElevationMapping.cpp), the process dies after ticking the ElevationMap box in Rviz to show the elevation map or using elevation map in external postprocessing pipeline (gridmap_filters_demo) without displaying the map in Rviz.

@duyipai what version of eigen3 did you upgrade to? I'm using version 3.3.7 (default for Ubuntu 20.04) which was released a month after your post:
_Resolved through using newer version of eigen3. Hope it helps others._

Thank you all for any help.