ANYbotics/elevation_mapping #41: NAN value for elevation map

feignedjesse2, 2017-05-23:
Hello,

I've been working on a project of navigation on rough terrain. I am using Freenect + RTABMAP with kinect to map the environment and do localization. I have published all the required topics to the elevation_mapping node. However when I use my launch file as below, I always get NAN values from the elevation mapping node. Why is this happening?

There are by the way no errors thrown by any nodes. The TF-tree is published too.

`<launch>
<node name="Elevation_Map" type="elevation_mapping" pkg="elevation_mapping" respawn="false" output="screen">
    <param name="point_cloud_topic" value="/rtabmap/mapData"/>
    <param name="robot_pose_topic" value="/rtabmap/odomwithcovariance"/>
    <param name="base_frame_id" value="/Robot_Base"/>
    <param name="track_point_frame_id" value="/Robot_Base"/>
    <param name="resolution" value="0.1"/>
</node>

</launch>`

Kind Regards,
Jesse

johndevera, 2017-11-28:
Hi were you able to solve this? I too am getting NAN. Also can you help me with the pose? I used:
<param name="robot_pose_topic" value="/odom/pose"/>
this contains the pose and covariance information. Is this correct? I get an error saying "Could not get pose information from the robot". 

Any help you can provide me would be much appreciated!

Thanks, 
John

saikrn112, 2018-04-11:
Hi, 
@johndevera  `could not get pose information from the robot` can be solved by publishing `/geometry_msgs/PoseWithCovarianceStamped` on the topic name `/pose`
@feignedjesse2  did you solve the problem