ANYbotics/elevation_mapping #51: Error: Could not get pose information from robot

johndevera, 2017-11-28:
Hi, 

I am getting an error where it is not detecting my pose. I sample it from /odom/pose. This contains the pose and covariance information. Not sure what else to do. Any help would be appreciated. 

<node pkg="elevation_mapping" type="elevation_mapping" name="elevation_mapping" respawn="false" output="screen">
    <param name="point_cloud_topic" value="/camera/depth/points"/>
    <param name="robot_pose_topic" value="/odom/pose"/>
    <param name="base_frame_id" value="/base_link"/>
    <param name="map_frame_id" value="/map"/>
    <param name="track_point_frame_id" value="/base_footprint"/>
    <param name="resolution" value="0.05"/>
  </node>

Old-Man-Reality, 2019-09-10:
Hello,do you solve the problem?Can you tell me ?Thank u!!

maximilianwulf, 2019-09-18:
@johndevera there is a minimal working example as PR #105, this should answer your question.

@Old-Man-Reality if you have specific questions, feel free to open a new ticket and we will assist you in the implementation. Will close this for now.

closed 2019-09-18 by maximilianwulf

Old-Man-Reality, 2019-09-20:
thank u so much,I will try it .