ANYbotics/elevation_mapping #53: Updating map in real-time

johndevera, 2017-12-08:
I'm having trouble getting the map to update as soon as it has scanned an area. I've modified the various updating and publishing parameters to 1 hz but when viewed in Rviz it appears to update at random times, almost 10-20 seconds after scanning an area. In your video with the 4 legged robot over the rubble scene, the map appears to be saving and updating in realtime. What can you suggest is a good way to do that?

wojiaojiao, 2018-03-14:
@johndevera do you solve it?i have issues same with you

Rayman, 2018-03-16:
Did you compile in release mode? This solved most of the issues for me
```
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release
```

maximilianwulf, 2019-09-18:
Closing because of missing follow up. Feel free to reopen.

closed 2019-09-18 by maximilianwulf