ANYbotics/elevation_mapping #55: Clear map service call causes EM to stop outputing maps

mauricefallon, 2017-12-17:
Running Anymal in simulation, Elevation mapper stops working (but doesn't crash) when a clear_map service call is made.

I ran the following with the latest Anymal release:
`roslaunch anymal_nav_sim sim.launch`
this starts to publish RealSense elevation maps.

Then a call to this service causes the elevation mapping node to stop publishing:
`rosservice call /elevation_mapping/clear_map`

Any tips on how to reproduce more minimally? (also seen when EM is configured to use actuated lidar input)

pfankhauser, 2018-01-30:
We experience the same issue, we'll look into this asap, sorry for the inconvenience.

remod, 2019-03-22:
This should have been fixed with https://github.com/ANYbotics/elevation_mapping/pull/92. I am closing the issue, please report if you experience it again.

closed 2019-03-22 by remod