ANYbotics/elevation_mapping #56: Question: What is the argument for choosing the pose interface?

rokusottervanger, 2018-01-05:
Elevation mapping now listens to a pose topic with message type geometry_msgs/PoseWithCovarianceStamped. What is the argument for choosing this? Another option would be to enable the user to provide the pose using a nav_msgs/Odometry (maybe as an alternative), as this also contains the pose and covariance. A nice perk of an Odometry interface is easy integration with [robot localization](http://wiki.ros.org/robot_localization).

draliii, 2018-01-07:
I haven't found a way to solve this, so I use a node to export PoseWithCovarianceStamped from Odometry. Feel free to use/modify: [pose_translator](https://github.com/draliii/pose_translator)

rob3a, 2018-02-16:
@draliii How did you do that?

rob3a, 2018-02-21:
Sorry @draliii do you know any translator to get that PoseWithCovarianceStamped from tf because gmapping puts odometry in tf

maximilianwulf, 2019-04-25:
Hi @rokusottervanger, at the moment we don't support nav_msgs/Odometry. There is no reason to not use it. If anyone wants to write a feature to support odometry messages feel free to submit a pull request. In the meantime, @draliii's node should do the job.

@rob3a [this node](https://github.com/ANYbotics/elevation_mapping/blob/feature/turtlebot3_demo/elevation_mapping_demos/scripts/tf_to_pose_publisher.py) will listen to a tf transform and publish it as a PoseWithCovarianceStamped message.