ANYbotics/elevation_mapping #57: Stereo camera parameter descriptions missing/unclear

rokusottervanger, 2018-01-25:
The stereo camera sensor model includes the following parameters:
 - p_1
 - p_2
 - p_3
 - p_4
 - p_5
 - lateral_factor
 - depth_to_disparity_factor

However, there is no explanation of what these parameters mean. Can anyone help me with this?

Mayankm96, 2018-04-11:
Any updates on this? I am trying to use Zed camera for elevation mapping and am stuck at this step as well.

peci1, 2018-10-26:
See https://ieeexplore.ieee.org/document/6375037/ .

maximilianwulf, 2019-04-18:
Hi @rokusottervanger, 
this [Ph.D. thesis](https://www.research-collection.ethz.ch/handle/20.500.11850/284254) describes the parameters in more detail compared to the paper. 

In the future, we will add a detailed description of the parameters. 

Regarding new camera sensors, the users have to measure the corresponding parameters by themselves.

MattDom, 2019-04-22:
Thank you for this additional source, @maximilianwulf. 

Unfortunately, I don't fully understand the parameters of the sensor model, yet. I've "reverse-engineered" (in other words, looked it  [here](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/src/sensor_processors/StereoSensorProcessor.cpp) up) this particular formula:
[image]

However, I asume, that the _depth-to-disparity_-factor is given by the focal length (_f_) times the baseline (_B_). Further, I denoted the disparity by _d_. _J(i)_ is the column and _I(i)_ the row, as it seems to me.

Because I didn't know what to do further, I plotted everthing with the provided parameters in [here](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/config/sensor_processors/aslam.yaml). For the disparity, I asumend a range from 30 to 180 (just for a first shot).
[image]

So, at first glance, it seems plausible so far. The devation increases with larger distances, as expected. 

The questions that remain:
- How to solve for _p1_ to _p5_? I could imagine an iterative solution, started with an educated guess...
- Where does the _240_ comes from? (It seems a bit magic, or does it refer to a specific camera resolution?)
- In the example [config file](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/config/sensor_processors/aslam.yaml) a related project work is referenced. I searched quite a long time, but couldn't find anything. It would be great if someone could send me this work.

MattDom, 2019-05-17:
Okay, here are some news! At first, forget what I've wrote on the post above.

So, let's have a look at new theory. The first mistake in the above post was not to respect the spatiality of an image. The variance is not constant over the whole image. Why am I thinking that? Have a look at the following image. The surface in front of the camera is perfectly perpendicular the to optical axis. So, the two points have an equal distance to the camera in z-direction. But, the euclidean distance is longer (D1<D2). For my understanding, this means a point that is farther from the camera is more prone to errors.

[image]

This theory fits very pretty well to the formula above. I used [these parameters](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/config/sensor_processors/aslam.yaml) for testing and generated an image with the dimension 640x480 and a constant disparity from 100 (just a first-shot). 

Here is the result:
[image]

Looks good at first glace. However, it doesn't tell us what are the parameters stand for. And last but not leat where this magic _240_ comes from.

The magic _240_ was my biggest question. I'm pretty sure that this is half the resolution in one direction (2*240=480 --> a common resolution). But why is it fixed in the source code? 

Next, the parameters. At first _p_1_ seems to be just a common offset for the whole variance all over the image. _p_2_ looks like a gain without any (direct) relation to the disparity and _p_5_ is the same with respect to the disparity?

```Matlab
p_1 =   0.03287; % Offset without depending on disparity?
p_2 =  -0.0001276; % common gain?
p_3 =   0.4850; % Kind of a shift?
p_4 =   320; % 399.1046; % vertical shift?
p_5 =   0.000006735; % disparity related gain?
```
_p_4_ can be a vertical shift or, in other words, the principal point? I adjusted it in to my image resoltion and it seems to represent the principal point. So, just _p_3_ is left. This seems to be bit tricky. It shifts the image horizontally with respect to the disparity.  Have a look at the image below.

[image]

But now, two questions are left. How to setup the measurement and select the parameters _p_1_, _p_2_, _p_3_ and _p_5_? 

**Every idea is welcome!**

arainbilal, 2019-08-05:
Hi @maximilianwulf 

@pfankhauser thesis you have referenced in this post referred to a student project report for more details about the sensor error model for the RealSense ZR300 on page 35, as follows:
Takahiro, M., Fankhauser, P., and Bjelonic, M. (2017). “Error Analysis of the RealSense
ZR300, Obstacle Removal for Elevation Mapping, and Swing Trajectory Planning for
ANYmal”. Student’s Project. ETH Zurich, Switzerland.
Unfortunately, this report is not accessible; I think should be accessible from ETH library. Not sure if @MattDom  has found an answer about what these parameters mean, looking for more details too. If possible please help. Thanks.

MattDom, 2019-10-29:
I found an interesting repo, submitted by @HannesKeller. This contains (https://github.com/HannesKeller/sensor_model.git) contains MATLAB code to generate the sensor model for a stereo camera, based on these mystic parameters. Actually, I don't have that much time to analyze this in deep...

Maybe Hannes will read this and share his work, mentioned in the code :smiley:

maximilianwulf, 2019-11-06:
@MattDom thank you for your input and the link to the repo. @HannesKeller [seems to be](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/include/elevation_mapping/sensor_processors/StereoSensorProcessor.hpp#L5) the original author of that sensor processor.

> And maybe Hannes will read this and share his work, mentioned in the code smiley.

Let's see. :smiley: 

@arainbilal does [this link](https://www.researchgate.net/publication/327393434_Perceptive_Locomotion_for_Legged_Robots_in_Rough_Terrain) work for you?

arainbilal, 2019-11-06:
Thanks @maximilianwulf. Yes it does. Although, I end up adding another sensor processor with in the elevation mapping framework using different parameters from my own tests, but it is interesting to review the work done by Nguyen et.al and then extended by @pfankhauser. Thanks.

maximilianwulf, 2019-11-06:
@arainbilal good to know. In what setup are you using elevation mapping?

arainbilal, 2019-11-06:
For terrain mapping in robotics applications and only for research purpose.

maximilianwulf, 2019-11-06:
@arainbilal thanks, I was just curious on what cool projects elevation map gets used.

xiandaoo, 2021-09-07:
@maximilianwulf  @MattDom I ran into a similar problem, and luckily I seemed to find the answer in an article. However, it is not known whether this error model is completely correct. Hope this answer can help those in need.
[article](https://www.sciencedirect.com/science/article/abs/pii/S0921889009000438)
[image]
[image]