ANYbotics/elevation_mapping #60: sensor_frame does not exist

aybakana, 2018-02-07:
I am using a VLP-16 lidar sensor.
These are the configurations for elevation_mapping node:
```
point_cloud_topic: "/velodyne_points"
map_frame_id: "world"
robot_base_frame_id: "top_plate_link"
robot_pose_with_covariance_topic: "/pose/gazebo"
robot_pose_cache_size: 100
track_point_frame_id: "top_plate_link"
track_point_x: 0.0
track_point_y: 0.0
track_point_z: 0.0
length_in_x: 2.5
length_in_y: 2.5
position_x: 0.0
position_y: 0.0
resolution: 0.01
min_variance: 0.000009
max_variance: 0.01
mahalanobis_distance_threshold: 2.5
multi_height_noise: 0.0000009
sensor_processor/type: laser
sensor_processor/min_radius: 0.018
sensor_processor/beam_angle: 0.0006
sensor_processor/beam_constant: 0.0015
```
And I get this outcomes:
```
[ WARN] [1518015771.987702851, 1340.760000000]: Visibility cleanup duration is too high (current rate is 0.163138).
[ INFO] [1518015771.988841483, 1340.760000000]: ElevationMap received a point cloud (17629 points) for elevation mapping.
[ERROR] [1518015775.070534283, 1341.760000000]: "sensor" passed to lookupTransform argument source_frame does not exist. 
[ERROR] [1518015775.070570182, 1341.760000000]: Point cloud could not be processed.
[ INFO] [1518015775.071199457, 1341.760000000]: ElevationMap received a point cloud (17629 points) for elevation mapping.

```
I am trying to convert the point cloud to elevation map. I saw that world frame is published with a higher frequency than other frames and I can not see elevation map on rviz. What can be the problem? What am I doing wrong?

aybakana, 2018-02-08:
After I looked at the SensorProcessorBase.cpp code, I saw that "sensor_frame_id" is "/sensor" as a default value. I added the following configuration:
`sensor_frame_id: "/velodyne"`
And also I added velodyne_VLP-16.yaml configuration file under elevation_mapping/config/sensor_processors in order to help others.

closed 2018-02-08 by aybakana

MuMuJun97, 2018-10-31:
> After I looked at the SensorProcessorBase.cpp code, I saw that "sensor_frame_id" is "/sensor" as a default value. I added the following configuration:
> `sensor_frame_id: "/velodyne"`
> And also I added [...]

I'm reading this code, thank for you job

Martygp95, 2019-09-20:
Hi! How did you calculate the parameters of beam angle and constant for VLP-16?