ANYbotics/elevation_mapping #61: How to get the pose from gmapping

jav3a, 2018-02-16:
I am using gmapping to explore an unknown map  but I need to create an elevation mapping also. I do not know how to get /pose (pose with covariance stamped) from  entropy of gmapping (Float64) or from odom of gazebo(odometry). Maybe i need to use the tf and odom to get the real pose?? Many thanks for any help.

reinzor, 2018-03-20:
The idea of the fuse step in the map update method is that it takes the covariances of the measurements into account and fuses this to a coherent map. You could just manually add a covariance matrix based on some gaussian noise that describes the pose uncertainty of your robot.

aybakana, 2018-03-21:
The script below listens tf and publishes posewithcovariancestamped -> 

[ground_truth_pose_publisher.py](https://github.com/ethz-asl/elevation_mapping/blob/master/elevation_mapping_demos/scripts/ground_truth_pose_publisher.py )

closed 2018-05-25 by rob3a

TT22TY, 2020-03-12:
> The script below listens tf and publishes posewithcovariancestamped
> 
> [ground_truth_pose_publisher.py](https://github.com/ethz-asl/elevation_mapping/blob/master/elevation_mapping_demos/scripts/ground_truth_pose_publisher.py)

@aybakana ，hello, the script you provide does not exist, could you please share it again, thank you very much~

maximilianwulf, 2020-03-17:
Hi @TT22TY, 

[tf_to_pose_publisher.py](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping_demos/scripts/tf_to_pose_publisher.py) in the demo package should do the same.