ANYbotics/elevation_mapping #74: Can the project support velodyne HDL-64?

bryanibit, 2018-07-06:
I have recorded rosbag including pointcloud2 from velodyne-64, /tf for frame_id between the vehicle and the world and PoseWithCovarianceStampd for world_frame_id. The How can I use the bag to have an elevation_mapping. 
Now the elevation_mapping is filled with nan. The following is the result:
```
[ INFO] [1530867972.623350455]: Grid map visualization node started.
[ INFO] [1530867972.625583814]: grid_map_visualizations: Configured visualization of type 'point_cloud' with name 'elevation_cloud'.
[ INFO] [1530867972.625621464]: grid_map_visualizations: Configured visualization of type 'point_cloud' with name 'upper_bound_cloud'.
[ INFO] [1530867972.625649608]: grid_map_visualizations: Configured visualization of type 'point_cloud' with name 'lower_bound_cloud'.
[ INFO] [1530867972.625865611]: grid_map_visualizations: Configured visualization of type 'map_region' with name 'map_region'.
[ INFO] [1530867972.628691825]: Grid map visualization initialized.
[ INFO] [1530867972.633750244]: Grid map visualization node started.
[ INFO] [1530867972.636420228]: grid_map_visualizations: Configured visualization of type 'point_cloud' with name 'elevation_cloud'.
[ INFO] [1530867972.636457491]: grid_map_visualizations: Configured visualization of type 'point_cloud' with name 'sigma_cloud'.
[ INFO] [1530867972.636695711]: grid_map_visualizations: Configured visualization of type 'map_region' with name 'map_region'.
[ INFO] [1530867972.639628355]: Grid map visualization initialized.
[ INFO] [1530867972.671558473]: Elevation mapping node started.
[ INFO] [1530867972.697715222]: Elevation map grid resized to 250 rows and 250 columns.
[ INFO] [1530867972.726483010]: Elevation mapping node initializing ... 
[ INFO] [1530867973.726852942]: Done.
[ INFO] [1530867973.738195697]: ElevationMap received a point cloud (110592 points) for elevation mapping.
[ INFO] [1530867983.521865152]: Raw map has been updated with a new point cloud in 5.122441 s.
[ INFO] [1530867983.535307001]: ElevationMap received a point cloud (110592 points) for elevation mapping.
[ INFO] [1530867984.018407364]: Elevation map has been fused in 9.291024 s.
[ INFO] [1530867993.337481506]: Raw map has been updated with a new point cloud in 5.124817 s.
[ INFO] [1530867993.340234993]: Visibility cleanup has been performed in 18.612875 s (0 points).
[ WARN] [1530867993.340339130]: Visibility cleanup duration is too high (current rate is 0.053726).

[ INFO] [1530867993.351025602]: ElevationMap received a point cloud (110592 points) for elevation mapping.
[ INFO] [1530867993.816907762]: Elevation map has been fused in 9.796082 s.
[ INFO] [1530868003.225202005]: Raw map has been updated with a new point cloud in 5.208442 s.
[ INFO] [1530868003.227891645]: Visibility cleanup has been performed in 9.887359 s (0 points).
[ WARN] [1530868003.227953766]: Visibility cleanup duration is too high (current rate is 0.101139).

[ INFO] [1530868003.236804103]: ElevationMap received a point cloud (110592 points) for elevation mapping.
[ INFO] [1530868003.708482737]: Elevation map has been fused in 9.890168 s.
[ INFO] [1530868013.035706870]: Raw map has been updated with a new point cloud in 5.109959 s.
[ INFO] [1530868013.038773528]: Visibility cleanup has been performed in 9.810674 s (0 points).
[ WARN] [1530868013.038901000]: Visibility cleanup duration is too high (current rate is 0.101930).

[ INFO] [1530868013.047897045]: ElevationMap received a point cloud (110592 points) for elevation mapping.
[ INFO] [1530868013.548942611]: Elevation map has been fused in 9.839041 s.
[ INFO] [1530868022.799122446]: Raw map has been updated with a new point cloud in 5.141613 s.
[ INFO] [1530868022.801794363]: Visibility cleanup has been performed in 9.762718 s (0 points).
```

hello-sea, 2019-03-25:
Hello, I have encountered the same problem with realsense (d435). What parameters do I need to adjust?

hello-sea, 2019-03-25:
Hello, I have encountered the same problem in the process of using realsense (d435). What parameters do I need to adjust?

maximilianwulf, 2019-04-23:
Hi @bryanibit,

where do you get the message that elevation_mapping is filled with nan? Your output indicates proper functionality. The only thing I see is that that it takes quite long. Did you compile it in Release mode?

LZMHIT, 2019-05-12:
@maximilianwulf 
How can  I compile it in  Release mode?

maximilianwulf, 2019-05-13:
Hi @LZMHIT, 

if you are building with `catkin_make` you can change the command to `catkin_make -DCMAKE_BUILD_TYPE=Release`.

If you are using `catkin build`, you can instead use `catkin build -DCMAKE_BUILD_TYPE=Release`.

I hope this helps.

LottieWang, 2019-07-16:
I record PointCloud, PoseWithCovarianceStampd and tf in one rosbag. When I run 
**rosbag play -k --clock bag1.bag**, 
the elevation mapping node give me errors 
**Request update with time stamp 1563307972, but time of last update was 1563312236. 
Updating process noise failed.**
How could I run elevation_mapping with a rosbag?

maximilianwulf, 2019-07-17:
Hi @LottieWang, 
Before you start `rosbag play`, did you make sure to also enter the command `rosparam set use_sim_time true`?

It seems that the elevation mapping is using the clock from your computer and not the one from the bag.
1563307972 -> 07/16/2019 @ 8:12pm
1563312236 -> 07/16/2019 @ 9:23pm

There is more than 1 hour of difference between the timestamps.

maximilianwulf, 2019-09-18:
Related issue #116.

maximilianwulf, 2019-09-18:
Closing now because of inactivity, feel free to reopen.

closed 2019-09-18 by maximilianwulf