ANYbotics/elevation_mapping #79: Missing dependency on 'grid_map_rviz_plugin'

marco-tranzatto, 2018-08-06:
Hi all,

after I cleaned my workspace and build it again I noticed that rviz could not visualize the elevation map anymore. I believe we have a missing dependency on "grid_map_rviz_plugin". I suggest we include it here. If that's okay with you  I'll open a PR for this repo.

What do you think @pfankhauser @remod?

pfankhauser, 2018-08-06:
I don't think that makes sense, since the elevation mapping might be used on the robot without RViz. The `grid_map_rviz_plugin` is set as dependency at `anymal_sim`. Does this make sense?

remod, 2018-08-06:
It is a dependency of anymal_b_navigation_ui, so it is built whenever you build an opc or an npc package.

marco-tranzatto, 2018-08-06:
ok, I see. On my PC I did not build any of them after I cleaned the WS, but just the replay package, which does not depend on neither of them, therefore the rviz plugin was not compiled. Thanks for the explanations.

closed 2018-08-06 by marco-tranzatto