ANYbotics/elevation_mapping #86: LookupTransform argument target_frame in tf2 frame_ids cannot be empty

BrunoEduardoCSantos, 2018-11-08:
Hi, 
I am publishing a point cloud from velodyne VLP-16 and run elevation_mapping_demo using the ground_truth yaml as follows: 

```
point_cloud_topic: "/velodyne_points"
map_frame_id: "map"
sensor_frame_id: "velodyne"
robot_base_frame_id: "map"
robot_pose_with_covariance_topic: "/ground_truth_pose"
robot_pose_cache_size: 500
track_point_frame_id: "map"
track_point_x: 0.0
track_point_y: 0.0
track_point_z: 0.0

```
As tf and pose I use a SLAM algorithm to this purpose. Although, the tf is tf2 type. 
I got the following error:

`[ERROR] [1541700130.680266861, 1455209754.533483445]: Invalid argument passed to lookupTransform argument target_frame in tf2 frame_ids cannot be empty
`
Can anyone point me out what am I doing wrong?
Thanks in advance,
Bruno

BrunoEduardoCSantos, 2018-11-09:
I checked the point cloud frame and I got "/camera_init" as supposed.
Although, I still got the following warning:
ElevationMap received a point cloud (3800 points) for elevation mapping.
```
FRAME/camera_init
Warning: Invalid argument passed to canTransform argument target_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.18/src/buffer_core.cpp
```

closed 2018-11-09 by BrunoEduardoCSantos

maximilianwulf, 2019-04-23:
This problem was solved with #102.