ANYbotics/elevation_mapping #93: Map updated without data from sensor

HUDiwan, 2019-02-21:
Hello,

I have two questions. First, how are the stereo camera parameters set. I tried to look for the project info that is cited, but since it is semester project I didn't find anything. I have tried to follow [https://ieeexplore.ieee.org/document/6375037/] to set up the parameters for ZED camera but I couldn't get it to work, there is already someone who posted issue for it but it has not been addressed. Second, I am getting 3 errors, and I have no clue how to fix. So far I have got this done:

Error:
[ WARN] [1550710410.053304272]: Elevation map is updated without data from the sensor.
[ERROR] [1550710410.053370440]: Could not get pose information from robot for time 1550710410.053353. Buffer empty?
[ERROR] [1550710410.053390845]: Updating process noise failed.

I looked at my rostopic and I know my pose is being published on /pose2D which is what I wrote for my parameter. The other two errors I am not sure how to fix. 
 
This is my yaml file for robot:

point_cloud_topic: "zed/point_cloud/cloud"
map_frame_id: "map"
robot_base_frame_id: "wheelchair_base"
robot_pose_with_covariance_topic: "/pose2D"
robot_pose_cache_size: 200
track_point_frame_id: "wheelchair_base"
track_point_x: 0.0
track_point_y: 0.0
track_point_z: 0.0

Any suggestions?
 
Thanks.

harmishhk, 2019-03-01:
The error also appears on 19.02 pre-release branch. Can this be fixed easily @maximilianwulf ?
```
[ERROR] [1551431988.491340618, 3.337000000]: Could not get pose information from robot for time 3.242000. Buffer empty?
[ERROR] [1551431988.492348315, 3.338000000]: Updating process noise failed.
```

maximilianwulf, 2019-03-11:
Hello @HUDiwan 

The pull request should remove the Error warnings in the beginning. 
During startup it can happen that point clouds with an old time stamp arrive and therefore no matching pose can be found. We added a waiting phase for a matching pair of point cloud and pose at the startup of the node.
I hope this helps.

closed 2019-03-12 by harmishhk