ANYbotics/elevation_mapping #95: Mapping the whole terrain

LZMHIT, 2019-03-07:
Dear Author,
    Your work is amazing, I do some config in my gazebo simulation and I launch simple_demo.launch for a robot car with odom and kinect which provide pointclout2_message, but the map I got is a local map in the sight of the kinect vision sensor, besides, the map use only part of the points from kinect, I do not know why, What is more, When I control the Car, the map generated before disappeared, I think it is the local map refreshed when a new points coming in, am I wright? So how can I get the whole terrain without the disappearing for the terrain generated from before points? Do I need to make an alignment with two consecutive  points from two frames?

[image]
This is the first pose and the terrain generated with elevation_mapping package,

[image]
and this is the second one.
As we can see, the two terrains are departed.

How can I make them emerge to get a whole terrain like yours like following picture?
[image]

Looking for your answer?
Thanks with sincerely!

remod, 2019-03-08:
Hi @LZMHIT 

You can increase the map size in the [simple demo config file](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping_demos/config/elevation_maps/simple_demo_map.yaml), by adapting length_in_x and length_in_y. The map is robot-centric, so when the robot moves, areas which are far away will disappear.

Cheers
Remo

LZMHIT, 2019-03-08:
> robot

Oh I see, thanks a lot, and how can I get the world-centric map without aeras disappearing?

pfankhauser, 2019-03-08:
You can set the point the map follows with the parameter:

`track_point_frame_id` (string, default: "/robot")

The elevation map is moved along with the robot following a track point. This is the id of the tf frame in which the track point is defined.

closed 2019-03-08 by pfankhauser

LZMHIT, 2019-03-08:
> You can set the point the map follows with the parameter:
> 
> `track_point_frame_id` (string, default: "/robot")
> 
> The elevation map is moved along with the robot following a track point. This is the id of the tf frame in which the track point is defined.

I did this before, I change it to "map", however it dosen't work. The areas still disappeared

JiajieLai, 2020-10-28:
Hi, mate, I suffer from the same problem, too. Could you please share some ideas to fix it?

maximilianwulf, 2020-10-28:
There is currently no setting to map everything.

But you can change the frame as explained. And also increase the lenght of the map. So if your whole terrain is 300mx300m make sure to adapt the parameter accordingly.