ANYbotics/elevation_mapping #99: There is a fault, invoking "make run_tests_elevation_map_msg run_tests_elevation_mapping -j12 -l12" failed, when I catkin_make run_tests_elevation_map_msg run_tests_elevation_mapping.Why ?

hello1future, 2019-03-25:


maximilianwulf, 2019-04-08:
Hi @hello1future, 
can you provide a screenshot of your console?

hello1future, 2019-04-08:
The fault has been resolved.
Now,A new problem has arisen
[image]

What should I do? Can you help me?I'm a new ros guy.
Thank you very much.

maximilianwulf, 2019-04-09:
This behavior is intended. The problem is that no depth and pose data gets generated in the example. The data needs to come from another source. Usually, it comes from
- a real robot platform,
- a [bag](http://wiki.ros.org/ROS/Tutorials/Recording%20and%20playing%20back%20data) that gets replayed, or
- a [simulation environment](http://wiki.ros.org/turtlebot_gazebo/Tutorials/indigo/Explore%20the%20Gazebo%20world), e.g. Gazebo.

Are you planning to use the software in simulation or with a real platform?

hello1future, 2019-04-09:
Thank you for your answer.
 We use it in a simulation environment, if it can meet our requirements, in the later, we can use it on a real platform.

maximilianwulf, 2019-04-10:
Okay, then you have to run the simulation together with elevation mapping.
Make sure that elevation mapping subscribes to your point cloud topic and pose topic. For the simple_demos.launch you can define the parameters in the file [simple_demo_robot.yaml](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping_demos/config/robots/simple_demo_robot.yaml).
I hope this helps.

hello1future, 2019-04-10:
Thank you very much.
I want to test elevation_mapping with Turtlebot3 simulation environment.  Feasible?

maximilianwulf, 2019-04-11:
I am not familiar with the Turtlebot3 simulation environment. 
If it has a pose, i.e. from odometry, and a point cloud, then it should work.

Would be interesting if you can share your experience.

hello1future, 2019-04-11:
What's the reason why this error? Can you help me?
[image]

maximilianwulf, 2019-04-16:
Elevation mapping is not subscribed to the correct topic name. You have to change it in this file:  https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping_demos/config/robots/simple_demo_robot.yaml
Update 'point_cloud_topic' and 'robot_pose_with_covariance_topic'.

While running the Turtlebot3 simulation, you can list the available ros topic names with the command
`rostopic list`

Search for the point cloud and pose topic.

maximilianwulf, 2019-04-25:
Hi @hello1future, this pull request #105 integrates elevation mapping in the turtlebot3 simulation. 
I hope this helps. 
Feel free to reopen the thread if you have any problems.

closed 2019-04-25 by maximilianwulf

infinity0323, 2020-01-08:
> The fault has been resolved.
> Now,A new problem has arisen
> [image]
> 
> What should I do? Can you help me?I'm a new ros guy.
> Thank you very much.

Hello.I have the same fault now.Can you tell me how you solved it?Thank you.

infinity0323, 2020-01-08:
> The fault has been resolved.
> Now,A new problem has arisen
> [image]
> 
> What should I do? Can you help me?I'm a new ros guy.
> Thank you very much.

my error: make run_tests_elevation_map_msg run_tests_elevation_mapping -j8 -l8

maximilianwulf, 2020-01-17:
> my error: make run_tests_elevation_map_msg run_tests_elevation_mapping -j8 -l8

In order to run elevation mapping you don't need to compile the unit tests.

What frame are you displaying?