ANYbotics/elevation_mapping #103: How Can I run this software in real time?

LZMHIT, 2019-04-20:
Dear Author, I have tried your work, it is amazing, however when I run it, the map grow up slowly, it couldn't refresh the map  in real time against with my robot moving. Could you please tell me why?

maximilianwulf, 2019-04-23:
Hi @LZMHIT, thank you for using elevation mapping.

In which frequency did it run on your system? It depends on the hardware and the settings. 
This pull request #96 discusses some possibilities to increase fps.

> And what is more ,can I get the whole map instead of the robot-centric map?

Do you mean that the map should constantly grow? The elevation map is always robot-centric.

LZMHIT, 2019-04-23:
> Hi @LZMHIT, thank you for using elevation mapping.
>
> In which frequency did it run on your system? It depends on the hardware and the settings.
> This pull request #96 discusses some possibilities to [...]
Yeah, that just what I mean, thank you for your reply, I have another question. Yesterday, I tried to use 
elevation mapping with velodyne 32, however it seems that it can not get a continuous terrain mapping, can I use the parameters like the struct_light?

pfankhauser, 2019-04-23:
@LZMHIT Did you make sure to build in "Release"? Although the elevation mapping was originally developed for a robot-centric formulation, you can also associate the map to a static frame and run it world-centric.

LZMHIT, 2019-04-23:
@pfankhauser I only build with catkin_make, and I got a world-centric map by changing the track_point_frame_id in starleth.yaml, however how can I fuse all the every data frame form the sensor like kinect together in to the map? Till now, I can only get the map from the newest sensor data, while it disappeared when the next frame come, could you please tell me how to fused them together?

maximilianwulf, 2019-04-23:
> > Yeah, that just what I mean, thank you for your reply, I have another question. Yesterday, I tried to use elevation mapping with velodyne 32, however it seems that it can not get a continuous terrain mapping, can I use the parameters like the struct_light?

You can use this [YAML file](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/config/sensor_processors/velodyne_HDL-32E.yaml) for Velodyne 32.

DLonng, 2021-11-29:
> Is there a yaml file for robosense-16？

maximilianwulf, 2021-11-29:
@DLonng, no, but feel free to make a PR to this repo if you tune a new configuration file.

DLonng, 2021-11-29:
> @DLonng, no, but feel free to make a PR to this repo if you tune a new configuration file.

ok, thanks