ANYbotics/elevation_mapping #104: How to make this work with velodyne32

Zhousiyuhit, 2019-04-23:
Dear Author, I am sorry to bother you again, I have tried to use Velodyne32 to your project, I tried "roslaunch velodyne_description example.launch" as a gazebo simulator. And I advertise the Pose with tf, however I get the result like this:
[image]
The result show that the map is separately, 
and my parameters is imitative of the struct_light parameters:
 
[image]
I am so confused with it, could you please tell me how to work with Velodyne32? Thanks sincerely!!

YoshuaNava, 2019-04-23:
Hi @Zhousiyuhit,
I think the main problem is that the tf trees of your robot, and the velodyne hdl32 are not connected correctly. I would suggest inspecting the transformations between all your sensors (in rviz, or using [tf tools](http://wiki.ros.org/tf/Debugging%20tools)).

Maybe you could use this as a reference on how to add a Velodyne sensor (in this case VLP-16) to your robot URDF? https://github.com/njkaiser/nu_jackal_autonav/blob/master/urdf/jackal.urdf.xacro#L233

maximilianwulf, 2019-04-23:
Hi @Zhousiyuhit,
As @YoshuaNava suggested, you have to check the tf tree. Once you figured out the frame id's you can add them to [this file](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping_demos/config/robots/simple_demo_robot.yaml) or any other in that folder. Make sure to update the _map_frame_id_, _robot_base_frame_id_ and _track_point_frame_id_.

maximilianwulf, 2019-04-23:
Furthermore, for the parameters, you can use [velodyne_HDL-32E.yaml](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/config/sensor_processors/velodyne_HDL-32E.yaml).

LZMHIT, 2019-04-23:
@maximilianwulf I am the partner of @Zhousiyuhit, Thank you for your answer, we have made a check, we think the tf tree is complete, and the frame_ids are all updated corresponding to the tf frame.  Frame velodyne2 is the sensor frame and velodyne2_base_link is its base frame.
[image]
[image]
Is there any wrong with our settings?Besides, your project is running successfully with out any fault. However there is no terrain in rviz. Do you have any example of velodyne32 shared? Thank you very much.

YoshuaNava, 2019-04-23:
Hi @LZMHIT 
From the above image I can see that you don't have any [world-fixed frame](https://www.ros.org/reps/rep-0105.html) in your tf tree. This is needed by elevation_mapping.

LZMHIT, 2019-04-23:
Hi @YoshuaNava 
I have the world-fixed frame, it is base_footprint,and in my setting the map_frame_id is just base_footprint
[image]

YoshuaNava, 2019-04-23:
Do you see any warnings from elevation_mapping in the console?

LZMHIT, 2019-04-23:
Hi @YoshuaNava 
Only the first time it started there is an warning, after that I didn't see any warnings or wrong 
[image]

maximilianwulf, 2019-04-23:
Your settings look correct so far. 

Can you check if the elevation mapping node is publishing an elevation mapping topic with `rostopic hz /topic_name`.

Furthermore, to see the elevation map in RVIZ you have to add the grid map plugin. You should be able to access it with the 'Add' button.

[image]

maximilianwulf, 2019-04-23:
Your console output looks fine. Elevation map is subscribing to the point cloud topic and is updating the map. You should be able to see it in RVIZ.

LZMHIT, 2019-04-23:
Hi @maximilianwulf 
Thank you for your advice, I have added it, in the following picture, I can see the beautiful map when I zoom in it, however it is a little and is not continuously, and the topic is not always nan
[image]
[image]

maximilianwulf, 2019-04-25:
Hi @LZMHIT, by looking at your console output it everything seems okay.
There is a new pull request #105  with an example that uses the turtlebot3 simulation. The settings are correct and it could you help integrate it into your solution.

brunoeducsantos, 2019-08-28:
Hi @maximilianwulf ,
I am trying to use ouster lidar 64 in elevation_mapping. My tf tree is:

[image]

And my configuration file is:

```
point_cloud_topic: "/os1_cloud_node/points"
map_frame_id: "odom"
robot_base_frame_id: "base_link"
sensor_frame_id: "os1_lidar"
robot_pose_with_covariance_topic: "/base_link_pose"
robot_pose_cache_size: 200
track_point_frame_id: "base_link"
track_point_x: 0.0
track_point_y: 0.0
track_point_z: 0.0
 
```

The following issue is coming up when running elevation_mapping node:

```
[ERROR] [1566985013.395180598, 1551442133.538437939]: Lookup would require extrapolation into the past.  Requested time 614.657018000 but the earliest data is at tim
e 1551442129.432109387, when looking up transform from frame [os1_lidar] to frame [odom]
[ERROR] [1566985013.395503711, 1551442133.538437939]: Point cloud could not be processed.
```

Any thoughts what could be happening?
Thanks,
Bruno

YoshuaNava, 2019-08-28:
Hi @brunoeducsantos,
It seems that your Lidar data is coming in with a really old timestamp: 614.657018000 seconds.  In Unix time, that is:
[image]
Source: https://www.unixtimestamp.com/index.php

From what I know, the Ouster driver uses ros::Time to stamp sensor data. And you seem to be running the Lidar on a PC with a date not properly set. I would recommend that you check the clock of all the PCs that will be interacting in your robot.

Please let me know if this helps.

Best,
Yoshua Nava

brunoeducsantos, 2019-08-29:
Thanks @YoshuaNava . I will check out if that solves my issue.

maximilianwulf, 2019-09-18:
Hi @brunoeducsantos, does it work now with your setup?

brunoeducsantos, 2019-09-29:
Hi @maximilianwulf, 
Sorry for the late response. I sorted out the issue by overriding the time stamp and republish it.
Thanks for the query.

maximilianwulf, 2019-09-30:
@brunoeducsantos, good to hear. 
I will close this issue, feel free to reopen if further assistance is needed.

closed 2019-09-30 by maximilianwulf