ANYbotics/elevation_mapping #109: Real time is a little bit bad

hello1future, 2019-07-04:
I use the parameters in the picture below to run this demo. Its real-time performance is a little poor. May I ask how to improve it or what parameters should be modified？
[image]

hello1future, 2019-07-04:
This is my computer configuration
[image]

maximilianwulf, 2019-07-29:
Hi @hello1future, 

What about frequency are you aiming for?

- [ ] Is it compiled in Release?

In general, a high update frequency of the map is only necessary if the scene exhibits fast dynamic changes. As long as these are low, the standard settings should be fine if the pose gets updated with a reasonable speed (400Hz). 

At the moment there are different parameters to increase the processing speed, but they need to be tuned for the special application. 

- In case of a static environment, use a rather large map (increase `length_in_x` and `lenght_in_y`). 
- In case of a dynamic environment, reduce the size of the map to the close environment and try to downsample the input point cloud. For instance, you can use the voxel grid filter, which downsamples the point cloud. Add the following two parameters to your config file: 
`sensor_processor/apply_voxelgrid_filter`
`sensor_processor/voxelgrid_filter_size`
And you can only use the raw map instead of the fused map.

itaouil, 2021-06-29:
Hi @maximilianwulf,

I am trying to get the best frequency for the raw elevation map as we are aiming to use the mapping in a very dynamic scenario.

Using the voxelgrid filter that you suggested in the answer above I managed to go from 15Hz to 28Hz (using a 0.1 as the filter size).

Would you have other suggestions in order to increase the output frequency of the raw elevation map as the bottleneck in the **PointcloudCallback** seems to be the point cloud processing (i.e. **sensorProcessor_->process**)?

I am also aware that a GPU based version was developed, but I guess this is not open-source. Right?

maximilianwulf, 2021-06-30:
In general, processing minimal required data will give the currently best performance. Maybe you also iterate through point cloud data outside of the elevation map? 

 If you further want to increase latency, you should as you mentioned profile the PointCloudCallback function and add parallelization. This would be up to you then. 

> I am also aware that a GPU based version was developed, but I guess this is not open-source. Right?

There is none that I am aware of. 

Best, 
Maximilian

itaouil, 2021-06-30:
Ok I see.

Thank you!

Best,
Ilyass

maximilianwulf, 2021-06-30:
Can we close this then? 

Best,
Maximilian

itaouil, 2021-06-30:
Yeah sure.

I am not the original author of the issue, but regarding my enquiry you can certainly close it :).

Best,
Ilyass

maximilianwulf, 2021-06-30:
Nice, feel free to re-open if there are further questions.

closed 2021-06-30 by maximilianwulf