ANYbotics/elevation_mapping #110: Is there a data set that can be used to test it?

jacketfffff, 2019-08-02:
We want to use the elevation map on our robot,and We hope We can use the data set to see the effect before actually using it...

maximilianwulf, 2019-08-02:
Hi @jacketfffff, not at the moment, but I think it would be possible. Maybe some other users want to share one dataset.

In the meantime, you can use [this](https://github.com/ANYbotics/elevation_mapping/pull/105) simple demo with turtlebot3 to evaluate it. Let me know if you are happy with it.

jacketfffff, 2019-08-02:
Thank you, sir,and I also hope to know the version of gazebo. My gazebo is 7.15.0,but the process always died...

maximilianwulf, 2019-09-18:
I am using gazebo 9.0.0 and it works on my machine. Did you make it run in the meantime?