ANYbotics/elevation_mapping #111: I want to put elevation_mapping to the web with ros3djs , what should i do?

BGMjie, 2019-08-02:
At the ros web tool page , there is some interface to show urdf and pointcloud. But there is no interface to show elevation_mapping . What should I do can make it possible ?

maximilianwulf, 2019-08-02:
Hi @BGMjie, the elevation map is visualized with the grid map rviz plugin in rviz. It can be found [here](https://github.com/ANYbotics/grid_map/tree/master/grid_map_rviz_plugin). 

I guess you can copy the behavior from it quite easily. It would be nice to have it in the grid map library. We would appreciate a PR for this in the grid map repo.

BGMjie, 2019-08-02:
Thanks for your reply ! I having know that just now form my colleague . 
[image]
This is my rviz setting . It visualize elevation map though the plugin as you says.
But I have looked through the ros3djs libary before , there is no support for visualize grid map . Is that need me to write a funtion achive my goal?

maximilianwulf, 2019-09-18:
Hi @BGMjie, yes you have to write code to visualize it in ros3djs. 
Did you do it in the meantime? If you struggle with some implementation, we can help out.

BGMjie, 2019-09-18:
Actually there is a package can transfrom grid map to pointcloud2 . Roslib3djs can show the later . The problem sloved . Thanks for answering!

closed 2019-09-18 by BGMjie