ANYbotics/elevation_mapping #116: Using elevation_mapping with velodyne HDL-32

Martygp95, 2019-09-18:
Hi again, I'm using Husky robot and the Velodyne HDL-32; I have an output in topic elevation_mapping/elevation_map, but the map visualised in Rviz is not consistent, there are like holes when detecting a wall.
[image]

I have the configuration files like this:
[image]
The topic output is like:
[image]

Considerations: The sensor is properly attached to the robot as well. I have compiled the code in Release, but the algorithm works really slow on rviz.

maximilianwulf, 2019-09-18:
Hi @Martygp95, did you move the robot around?
Holes in the elevation map come from non-existent data in that area. Especially the Velodyne point cloud is not very dense. By driving around the missing spots should be filled at some point. 
Does this answer your question?

> but the algorithm works really slow on rviz.

Did you compile the rviz plugin form grid_map in release mode?
The plugin in its current state is not optimal but for us, it works.

Martygp95, 2019-09-18:
okey i'll try later! thank you so much!

Martygp95, 2019-09-18:
And with release mode I just cloned the repo of grid_map and did catkin_make -DCMAKE_BUILD_TYPE=Release

Martygp95, 2019-09-19:
I did few changes and it worked well! I changed elevation map parameters as it follows:

length_in_x: 5.0
length_in_y: 5.0
position_x: 0.0
position_y: 0.0
resolution: 0.08
min_variance: 0.00009
max_variance: 0.01
mahalanobis_distance_threshold: 2.5
multi_height_noise: 0.00002

So i put less resolution to the map, I tried to move the robot and it worked perfectly.

maximilianwulf, 2019-09-19:
Great to hear. Would you mind sharing a screenshot of the elevation map in this thread?

Martygp95, 2019-09-19:
[image]

maximilianwulf, 2019-09-19:
Thank you very much. I will close this issue now, feel free to reopen if you have further questions.

closed 2019-09-19 by maximilianwulf