ANYbotics/elevation_mapping #119: Output as a 2D Occupancy Grid

Dev727, 2019-10-28:
Hello,

I have managed to get a decent live reading of data using a Velodyne Lidar. 

The output is currently a standard Grid Map (based on the Simple Demo), but I was hoping that I would be able to adjust the output type similar to how Grid Mapping works.

Any help would be appreciated.
Thanks

maximilianwulf, 2019-11-06:
Hi @Dev727, 

> The output is currently a standard Grid Map (based on the Simple Demo), but I was hoping that I would be able to adjust the output type similar to how Grid Mapping works.

In what way do you want to adjust the output type?
If you are interested to convert the grid_map to a costmap2d object you have to write the transformation for yourself for now. At the moment grid_map only supports the conversion to a costmap2d.

Dev727, 2019-11-06:
> At the moment grid_map only supports the conversion to a costmap2d.
This should be exactly what I'm looking for. Is there a simple way to convert the produced Grid Map to a Costmap2D using the Simple Demo? Would it require editing the .yaml file or actually altering any of the actual code?

Also, thanks for the response, @maximilianwulf.

maximilianwulf, 2019-11-06:
@Dev727 at the moment you have to write your own code to get the behavior. 

You can find inspiration to do it in the [costmapt2GridMap converter](https://github.com/ANYbotics/grid_map/blob/master/grid_map_costmap_2d/include/grid_map_costmap_2d/Costmap2DConverter.hpp) class.