ANYbotics/elevation_mapping #120: catkin build error

Levi568, 2019-11-15:
**I successfully build the dependencies including kindr_ros and grid_map and make install the kindr repo.** 

The kindrConfig.cmake is installed :
-- Installing: /usr/local/share/kindr/cmake/kindrConfig.cmake

**But when I catkin build the elevation_mapping repo, it shows the following error :**

Errors     << elevation_mapping:cmake /home/wei/catkin_ws/logs/elevation_mapping/build.cmake.005.log
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "kindr" with any of
  the following names:

    kindrConfig.cmake
    kindr-config.cmake

  Add the installation prefix of "kindr" to CMAKE_PREFIX_PATH or set
  "kindr_DIR" to a directory containing one of the above files.  If "kindr"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:7 (find_package)

**Any comments would be helpful, thanks!**

maximilianwulf, 2019-11-18:
Hi @Levi568, it can't find the kindr package in the installed path (/usr/local/share/kindr/cmake/). This can have several reasons. In which workspace are you building `kindr_ros`?

Levi568, 2019-11-18:
@maximilianwulf I successfully built the kindr_ros package in catkin workspace.

maximilianwulf, 2019-11-19:
It looks like that the paths are not set correctly. Can you copy-paste the output of `catkin config`?

Levi568, 2019-11-19:
--------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/wei/catkin_ws
--------------------------------------------------------
Build Space:        [exists] /home/wei/catkin_ws/build
Devel Space:        [exists] /home/wei/catkin_ws/devel
Install Space:      [unused] /home/wei/catkin_ws/install
Log Space:          [exists] /home/wei/catkin_ws/logs
Source Space:       [exists] /home/wei/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------

maximilianwulf, 2019-11-20:
Is the `kindr_ros` package located inside ` /home/wei/catkin_ws/src`?

Levi568, 2019-11-20:
Yes, and I built it successfully.

Old-Man-Reality, 2020-07-19:
Hi! I meet the same problem,Did you solve it?Can you tall me how to solve it?

Old-Man-Reality, 2020-07-19:
@Levi568 @maximilianwulf

Levi568, 2020-07-19:
@Old-Man-Reality Unfortunately, I didn't end up solving this issue.

Old-Man-Reality, 2020-07-19:
I solve it just now,you can take the kind_ros  package to the elevation_mapping_ws/src,then elevation_mapping package and the kindr_ros package int the same src,finally you use catkin_make.the problem is solved.

Old-Man-Reality, 2020-07-19:
@Levi568

NmBoyd, 2022-07-11:
Changing the CMakeList to find kindr using `find_package(kindr REQUIRED)` stabilized `catkin build` for me since it's an external, non-ros, library.

maximilianwulf, 2022-07-12:
@NmBoyd what is the general workflow when git cloning this repo? 

I assume that as an open-source use you have to install the package dependencies with http://wiki.ros.org/rosdep?

skywalker-dell, 2022-11-04:
I encountered this problem when I am using `catkin build` to compile code. However `catkin_make` works fine.
> sudo ln -s /usr/local/share/kindr /opt/ros/<ros_version>/share/kindr
It works for me.

naveenkumar2208, 2023-04-08:
Hi, The solution is create a new workspace. Add kinder_ros package and https://github.com/ANYbotics/message_logger.git package and then build it using catkin_make. It worked for me.

amgaber95, 2023-04-27:
> Changing the CMakeList to find kindr using `find_package(kindr REQUIRED)` stabilized `catkin build` for me since it's an external, non-ros, library.

This solve the issue, since I am also using `catkin build`