ANYbotics/elevation_mapping #124: How to speed up visibility cleanup?

brent7984, 2019-12-06:
Hi, I have been tasked to use ANYbotics Elevation Mapping to do a traversability analysis of a surface. I had to find out whether a robot can move up a slope considering obstacles hanging from the ceiling. I managed to get this working, but I am encountering speed issues with the visibility cleanup.

I am using an Asus Xtion RGBD camera. I downsampled the pointcloud, to work with only 3000 points. I limited the Elevation_Map_Region to 3.5x3.5m. and I am using Ubuntu 18.04 on an Intel Core i5-8350U. The cell resolution was set to "0.12" and I changed the uncertaintyFactor from "2.486 to 0.01"

The elevation map is fusing in approx 0.5s. The Visibility cleanup says that its operating at approx 2 Hz but it takes over 30 seconds to completely remove an object from the map.

Is there any way for me to speed up the visibility cleanup. I tried adjusting the variables shown in the github page, but it did not speed it up.

Also is there any way for me to speed up the fusing times. I've noticed that sometimes I receive 2 pointclouds and then the fusing takes place, is there anyway for me to fused the map after each pointcloud.

With regards
Brenton

maximilianwulf, 2019-12-19:
Hey @brent7984, 

Fusing and visibility cleanup take a lot of time. 
Depending on your application and the sensor quality you could just not use fusing and the visibility cleanup and subscribe to the raw elevation map topic directly. Did you have a look at the quality of it?

How noisy is the Asus Xtion RGBD camera?

If it is really necessary you might have to optimise the code.

brent7984, 2020-01-09:
Thank you, subscribing to the raw elevation map did help.

brent7984, 2020-01-09:
It still would be useful though, to speed up visibility cleanup for future projects.

maximilianwulf, 2020-01-10:
@brent7984 thanks, good to know. 

> It still would be useful though, to speed up visibility cleanup for future projects.

That is true, MRs are always welcome.

Will close this for now, feel free to reopen.

closed 2020-01-10 by maximilianwulf