ANYbotics/elevation_mapping #126: unit tests error

infinity0323, 2020-01-08:
Hi.
**When I run the unit tests with**

catkin_make run_tests_elevation_map_msg run_tests_elevation_mapping

 **it shows the following error :**
 no rules to make target “run_tests_elevation_map_msg”

**Why? I have already built 'catkin_ws' correctly.**

maximilianwulf, 2020-01-17:
Hi @infinity0323, in order to use the package, you don't need to run the unit tests. 
What is your plan with elevation mapping?

infinity0323, 2020-03-27:
> Hi @infinity0323, in order to use the package, you don't need to run the unit tests.
> What is your plan with elevation mapping?

I want to use it to do the path planning of UAV.
I don't know if it is appropriate.
Still learning. 
Do you have any suggestions？>_<

maximilianwulf, 2020-06-26:
Yes, check these papers: 

[Active autonomous aerial exploration for ground robot path planning](https://www.zora.uzh.ch/id/eprint/138916/1/RAL16_Delmerico.pdf)

[Collaborative localization of aerial and ground robots through elevation maps](https://www.zora.uzh.ch/id/eprint/127901/1/SSRR16_Kaeslin.pdf)

closed 2020-06-26 by maximilianwulf