ANYbotics/elevation_mapping #128: Problem running demos

pauljurczak, 2020-01-09:
I'm having problem running `elevation_mapping_demos`. Rviz starts with an empty grid and `elevation_mapping` dies. `elevation_mapping-2*.log` file is not created.

```
paul@paul-desktop:~/catkin_ws$ roslaunch elevation_mapping_demos simple_demo.launch
... logging to /home/paul/.ros/log/9c7e297a-32b4-11ea-84c5-04d9f5cbc433/roslaunch-paul-desktop-10117.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://paul-desktop:39693/

SUMMARY
========

PARAMETERS
 * /elevation_map_fused_visualization/grid_map_topic: /elevation_mappin...
 * /elevation_map_fused_visualization/grid_map_visualizations: [{'type': 'point_...
 * /elevation_map_raw_visualization/grid_map_topic: /elevation_mappin...
 * /elevation_map_raw_visualization/grid_map_visualizations: [{'type': 'point_...
 * /elevation_mapping/length_in_x: 2.5
 * /elevation_mapping/length_in_y: 2.5
 * /elevation_mapping/mahalanobis_distance_threshold: 2.5
 * /elevation_mapping/map_frame_id: map
 * /elevation_mapping/max_variance: 0.01
 * /elevation_mapping/min_variance: 9e-06
 * /elevation_mapping/multi_height_noise: 9e-07
 * /elevation_mapping/point_cloud_topic: /points
 * /elevation_mapping/position_x: 0.0
 * /elevation_mapping/position_y: 0.0
 * /elevation_mapping/resolution: 0.01
 * /elevation_mapping/robot_base_frame_id: base
 * /elevation_mapping/robot_pose_cache_size: 200
 * /elevation_mapping/robot_pose_with_covariance_topic: /pose
 * /elevation_mapping/sensor_processor/cutoff_max_depth: 3.0
 * /elevation_mapping/sensor_processor/cutoff_min_depth: 0.35
 * /elevation_mapping/sensor_processor/lateral_factor: 0.001376915
 * /elevation_mapping/sensor_processor/normal_factor_a: 0.0012
 * /elevation_mapping/sensor_processor/normal_factor_b: 0.0019
 * /elevation_mapping/sensor_processor/normal_factor_c: 0.4
 * /elevation_mapping/sensor_processor/normal_factor_d: 0
 * /elevation_mapping/sensor_processor/normal_factor_e: 1
 * /elevation_mapping/sensor_processor/type: structured_light
 * /elevation_mapping/track_point_frame_id: base
 * /elevation_mapping/track_point_x: 0.0
 * /elevation_mapping/track_point_y: 0.0
 * /elevation_mapping/track_point_z: 0.0
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    elevation_map_fused_visualization (grid_map_visualization/grid_map_visualization)
    elevation_map_raw_visualization (grid_map_visualization/grid_map_visualization)
    elevation_mapping (elevation_mapping/elevation_mapping)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [10127]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9c7e297a-32b4-11ea-84c5-04d9f5cbc433
process[rosout-1]: started with pid [10139]
started core service [/rosout]
process[elevation_mapping-2]: started with pid [10142]
process[elevation_map_fused_visualization-3]: started with pid [10143]
process[elevation_map_raw_visualization-4]: started with pid [10144]
process[rviz-5]: started with pid [10145]
[ INFO] [1578556189.986243911]: Grid map visualization node started.
[ INFO] [1578556189.992272689]: Grid map visualization node started.
[ INFO] [1578556190.004026397]: grid_map_visualizations: Configured visualization of type 'point_cloud' with name 'elevation_cloud'.
[ INFO] [1578556190.004129650]: grid_map_visualizations: Configured visualization of type 'point_cloud' with name 'sigma_cloud'.
[ INFO] [1578556190.004208307]: grid_map_visualizations: Configured visualization of type 'map_region' with name 'map_region'.
[ INFO] [1578556190.009195044]: grid_map_visualizations: Configured visualization of type 'point_cloud' with name 'elevation_cloud'.
[ INFO] [1578556190.009295780]: grid_map_visualizations: Configured visualization of type 'point_cloud' with name 'upper_bound_cloud'.
[ INFO] [1578556190.009341042]: grid_map_visualizations: Configured visualization of type 'point_cloud' with name 'lower_bound_cloud'.
[ INFO] [1578556190.009395004]: grid_map_visualizations: Configured visualization of type 'map_region' with name 'map_region'.
[ INFO] [1578556190.013945361]: Grid map visualization initialized.
[ INFO] [1578556190.022925297]: Grid map visualization initialized.
[ INFO] [1578556190.025924749]: Elevation mapping node started.
[ INFO] [1578556190.034921513]: Elevation map grid resized to 250 rows and 250 columns.
[ INFO] [1578556190.049859028]: Elevation mapping node initializing ... 
[elevation_mapping-2] process has died [pid 10142, exit code -11, cmd /home/paul/catkin_ws/devel/lib/elevation_mapping/elevation_mapping __name:=elevation_mapping __log:=/home/paul/.ros/log/9c7e297a-32b4-11ea-84c5-04d9f5cbc433/elevation_mapping-2.log].
log file: /home/paul/.ros/log/9c7e297a-32b4-11ea-84c5-04d9f5cbc433/elevation_mapping-2*.log
[rviz-5] process has finished cleanly
log file: /home/paul/.ros/log/9c7e297a-32b4-11ea-84c5-04d9f5cbc433/rviz-5*.log
^C[elevation_map_raw_visualization-4] killing on exit
[elevation_map_fused_visualization-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

maximilianwulf, 2020-01-17:
Thanks for the report. I will look into it. 

Did you build in release mode?

pauljurczak, 2020-01-18:
Release mode.

maximilianwulf, 2020-01-28:
Hey @pauljurczak, 

The problem with the demos is that they require a running state estimation system of a robot that provides a pose with a covariance. 

We just merged in the [turtlebot demo](https://github.com/ANYbotics/elevation_mapping#turtlebot3-waffle-simulation). There you should be able to easily test it. Let me know if it works for you.

Regarding your log output, I could not recreate the same error message, it starts spinning and waits for the first pose.

Bibbidi-Babbidi-Boo, 2020-08-24:
Hi @pauljurczak , 

Does the code work for you now. I am facing the same problem

pauljurczak, 2020-08-24:
@Bibbidi-Babbidi-Boo I have not tried. I'm busy with other projects.