ANYbotics/elevation_mapping #129: Running the clear_map service multiple times causes the application to freeze

ashimasharaph, 2020-01-09:
I can use the service to clear the map. However, if I run the service multiple times the application freezes. This usually happens after the 3rd or 4th time the service is called, but sometimes it can be called even 10 times without any issues.

maximilianwulf, 2020-01-17:
Hey @ashimasharaph, we are aware of that problem. If you have a reliable way to trigger it we would be very happy. Since we were not really able to debug it yet.

brent7984, 2020-01-22:
Hi, I am working on the same project with ashimasharaph, we managed to solve this issue by commenting out the mutex code shown below........ We then called a script to call the clear service every 1 second. It seems to work fine. Should we expect any issues, as a result of us commenting out this line of code in ElevationMap.cpp.

bool ElevationMap::clear()
{
  boost::recursive_mutex::scoped_lock scopedLockForRawData(rawMapMutex_);
  //~ boost::recursive_mutex::scoped_lock scopedLockForFusedData(fusedMapMutex_);
  rawMap_.clearAll();
  rawMap_.resetTimestamp();
  fusedMap_.clearAll();
  fusedMap_.resetTimestamp();
  return true;
}

brent7984, 2020-01-22:
I've noticed that it crashes when both of the mutex codes are active. When one is commented out, the clear function seems to work fine when called iteratively. It also performs better when the "scopedLockForRawData" is commented out as compared to "scopedLockForFusedData" being commented out

maximilianwulf, 2020-02-21:
Hi @ashimasharaph and @brent7984, 

We did internal investigations and it turned out as @brent7984 mentioned that a deadlock can happen. The clear_map service call deadlock should no be prevented by the newest update. 

Thank you for your help and please let us know if there are further inconsistencies.

closed 2020-02-21 by maximilianwulf