ANYbotics/elevation_mapping #130: ZED camera parameters

blablebli-robots, 2020-01-16:
Hello, great package!

Has anyone had any success with the zed camera? If so, could you please share any info on your experience. I am considering this camera for use with elevation package.

Cheers!

maximilianwulf, 2020-01-17:
Hi @blablebli-robots, the only difference considering other depth sensors with respect to elevation mapping is the noise parameters of the camera. 

In general, you should be good to go.

blablebli-robots, 2020-01-17:
Thank you @maximilianwulf for your prompt response.

erinline, 2020-05-04:
@blablebli-robots I am also trying to use the ZED stereo camera with this package, what parameter values did you end up using?

blablebli-robots, 2020-05-20:
Hello @erinline, I gave up with the ZED stereo camera, due to noise and the intended application I decided going with a different setup.

maximilianwulf, 2020-06-26:
HI @blablebli-robots, could you share more insights into the observed noise of the ZED stereo cameras?

blablebli-robots, 2020-12-22:
Hey @maximilianwulf, the application required detection of smaller obstacles/depths on the floor. The ZED did not allow to resolve them as the upper and lower bounds were too far apart, don't recall exactly by how much. It was also difficult to track the floor in several lighting conditions and missed entire non featureless floor patches consistently.

maximilianwulf, 2021-01-06:
@blablebli-robots thank you for your reply. I will close this issue for now. Feel free to re-open.