ANYbotics/elevation_mapping #132: about continuous cleanup

pxlong, 2020-03-23:
Hi, thx for sharing this great mapping tool.

I have found that if I set enable_continuous_cleaup to true, it will not publish elevation map, I can not see the GridMap in rviz. 
I am using Livox Lidar (mid-100) to publish point cloud with 10Hz.

pxlong, 2020-04-26:
I have found that if I set enable_continuous_cleaup to true, it will be stuck at the **`map_.clear();`**

maximilianwulf, 2020-06-26:
Hi @pxlong, 

> I want the elevation map is always created from the latest point cloud. How to set the configuration to meet this requirement?

`enable_continuous_cleaup` is the correct option for this. `map.clear()` can be slow, but it should never get stuck. If you don't need fusing of the elevation map you can turn it off. In this case the `map.clear()` function should be faster.

closed 2020-06-26 by maximilianwulf