ANYbotics/elevation_mapping #135: Want to add 'occupancy' layer

ankitvora7, 2020-05-13:
Hi, I have an application where I want to add 'occupancy_2d' as a layer along with the existing layers like elevation, variance etc. What is the best way to go about it ? I have a couple of options but not sure which one would be the best/quickest

1. Write a new package that subscribes to the grid map that the elevation_mapping package publishes. Add my layer to it and republish it in a different topic.
2. Write a new package which does something similar to elevation_mapping for the elevation layer and then add my stuff for the occupancy_2d layer and publish it in a topic.

I want to go with the 2nd option but seems like a lot of duplication of code from the elevation_mapping package. Any thoughts?

maximilianwulf, 2020-06-26:
Hi @ankitvora7, 

I would propose to go with 

> Write a new package that subscribes to the grid map that the elevation_mapping package publishes. Add my layer to it and republish it in a different topic.

I guess it makes more sense to join the elevation and occupancy grid in a new node without rewriting `elevation_mapping`. 

I think you can get inspiration from this code: https://github.com/ANYbotics/grid_map/tree/master/grid_map_costmap_2d

maximilianwulf, 2020-12-21:
Closing this one, feel free to re-open.

closed 2020-12-21 by maximilianwulf