ANYbotics/elevation_mapping #136: debug issue

dringakn, 2020-05-21:
Dear ANYbotics team,
Thank you sharing such a great package within the robotics community!
I have been using elevation mapping, however, I am facing the elevation mapping node crash when I set the node logging level to debug. Here is the debug output:

[DEBUG] [1590069916.836895896, 1589904836.722827935]: Requested to fuse an area of the elevation map with center at (6.967468, -3.480633) and side lengths (50.000000, 50.000000)
[DEBUG] [1590069916.836922000, 1589904836.722827935]: Fusing elevation map...
[DEBUG] [1590069916.966533748, 1589904836.984662168]: Elevation map is running visibility cleanup.
terminate called after throwing an instance of 'std::runtime_error'564956               
  what():  Time is out of dual 32-bit range

Could you please suggest the possible solution to the problem?

Regards,
Ahmad

maximilianwulf, 2020-06-26:
Hi @dringakn, 

> what(): Time is out of dual 32-bit range

This looks like a timing issue. What clock mode are you running?
And you can also point to where the code is crashing?

maximilianwulf, 2020-12-21:
Hey @dringakn, do you still have the issue?