ANYbotics/elevation_mapping #137: update the global map

EricWang1hitsz, 2020-07-27:
Hi, thanks for your amazing work! 
Recently, I'm working about path planning using elevation map. I firstly build a global elevation map and then update global map using local elevation map when the robot track the path.  I using the data of the local map to replace those of the global map. 
```
        gridMapGlobal = initialTraversabilityMap_;
        GridMap gridMapLocal;
        gridMapLocal = localTraversabilityMap_;
        for(grid_map::GridMapIterator iterator(gridMapLocal); !iterator.isPastEnd(); ++iterator)
        {
            Index index;
            index = *iterator;
            //ROS_WARN("Starting fusing map");
            auto& traversability = gridMapGlobal.at("traversability", index);
            auto& elevation = gridMapGlobal.at("elevation", index);
            //ROS_WARN("Get traversability layer of the fused map");
            if(!gridMapLocal.isValid(index, "traversability"))
                continue; // ignore cells without data in local map
            ROS_WARN_STREAM_ONCE("Started Index: " << std::endl << index << std::endl);
            traversability = gridMapLocal.at("traversability", index);
            elevation = gridMapLocal.at("elevation", index);
            ROS_WARN_ONCE("Get traversability layer of the initial map");
        }
        // fuse finish
        fusedTraversabilityMap_ = gridMapGlobal;
```

But I find that the frequency is slow, so I want to ask some good advice about that work? Thanks!

maximilianwulf, 2020-07-28:
Hi @EricWang1hitsz, 

> But I find that the frequency is slow, so I want to ask some good advice about that work?

What is your current frequency? Did you build in `Debug` or `Release` mode? How big is your grid map?

In the [readme](https://github.com/ANYbotics/grid_map#iterators) you find best practices about iterating. 

```
Note: For maximum efficiency when using iterators, it is recommended to locally store direct access to the data layers of the grid map with grid_map::Matrix& data = map["layer"] outside the for loop:
```

maximilianwulf, 2020-12-21:
Closing this one, feel free to re-open.

closed 2020-12-21 by maximilianwulf

aimiliosnrobotics, 2024-10-24:
@EricWang1hitsz @maximilianwulf  Hello, may i ask how you managed to create a global elevation map? Since elevation map is robot-centric (doesnt keep information that go outside of the map) and is of fixed size and seems to become very computationaly expensive when making it of huge size. Can we somehow create a grid map of dynamic size and will it work in terms of computational feasibility? Then i wouldnt update the global map, but instead have then the smaller robot-centric occupancy grid coming from the "small" elevation map work for the local planner whereas the global planner would plan on the occupancy grid provided by the global elevation map. So i would be very interested in learning how you managed to create a global elevation (grid) map. Thank you very much in advance!