ANYbotics/elevation_mapping #139: Elevation Mappiing node dies even in demo

Bibbidi-Babbidi-Boo, 2020-08-24:
Hi,

So currently I'm trying to use the elevation mapping package with the turtlebot demos as given in the readme page of the repo.
I have the following packages in my catkin_ws:

- eigen
- elevation_ mapping
- grid_map
- kindr_ros
- pcl
- perception_pcl
- turtlebot3

I use the `catkin_make -DCMAKE_BUILD_TYPE=Release` command to build the workspace.

There is no problem in building, but as soon as I run `roslaunch elevation_mapping_demos turtlesim3_waffle_demo.launch` I get a process has died message, and only the rviz opens. But when I run the rqt_graph, I see that the elevation mapping node crashes

[image]

I have tried the issues #120 #85 #77 #81 and #49.
Any help would be greatly appreciated

Thanks

maximilianwulf, 2020-08-24:
Hey @Bibbidi-Babbidi-Boo I can not confirm the issue on my setup. 

Are you using ubuntu 18.04 and ROS melodic?

Bibbidi-Babbidi-Boo, 2020-08-24:
Yes, I'm using ROS Melodic with Ubuntu 18.04

Bibbidi-Babbidi-Boo, 2020-08-25:
Hi @maximilianwulf were you able to replicate the issue?

maximilianwulf, 2020-08-25:
Hey @Bibbidi-Babbidi-Boo, no not on my machine. Can you build it in debug and show the backtrace, that is the only possibility I see to debug it.

Bibbidi-Babbidi-Boo, 2020-08-25:
Will do, can you tell me how to get the backtrace.

Bibbidi-Babbidi-Boo, 2020-08-25:
And if possible can you also tell me the versions of all packages you use

maximilianwulf, 2020-08-25:
Attach a gdb instance to your launch file: http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

And then as soon as you run into the segmentation fault enter `bt`. https://sourceware.org/gdb/onlinedocs/gdb/Backtrace.html

Bibbidi-Babbidi-Boo, 2020-08-25:
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7fffd3ba4700 (LWP 3045)]
[New Thread 0x7fffd33a3700 (LWP 3046)]
[New Thread 0x7fffd2ba2700 (LWP 3047)]
[New Thread 0x7fffd23a1700 (LWP 3048)]
[New Thread 0x7fffd1ba0700 (LWP 3049)]
[ INFO] [1598352855.480022854]: Elevation mapping node started.
[ INFO] [1598352855.503753681]: Elevation map grid resized to 60 rows and 60 columns.

Thread 1 "elevation_mappi" received signal SIGSEGV, Segmentation fault.
0x00007ffff7badd67 in elevation_mapping::SensorProcessorBase::SensorProcessorBase(ros::NodeHandle&, tf::TransformListener&) ()
   from /home/rishab/catkin_ws/devel/lib/libelevation_mapping_library.so
(gdb) bt
#0  0x00007ffff7badd67 in elevation_mapping::SensorProcessorBase::SensorProcessorBase(ros::NodeHandle&, tf::TransformListener&) ()
   from /home/rishab/catkin_ws/devel/lib/libelevation_mapping_library.so
#1  0x00007ffff7bbc7f9 in elevation_mapping::PerfectSensorProcessor::PerfectSensorProcessor(ros::NodeHandle&, tf::TransformListener&) ()
   from /home/rishab/catkin_ws/devel/lib/libelevation_mapping_library.so
#2  0x00007ffff7b738a7 in elevation_mapping::ElevationMapping::readParameters()
    () from /home/rishab/catkin_ws/devel/lib/libelevation_mapping_library.so
#3  0x00007ffff7b794a9 in elevation_mapping::ElevationMapping::ElevationMapping(ros::NodeHandle&) ()
   from /home/rishab/catkin_ws/devel/lib/libelevation_mapping_library.so
#4  0x000055555555b7c3 in main ()

Here it is

maximilianwulf, 2020-08-25:
Can you next go to frame 0 with the command `f 0` and then enter `list` to show at which position the segmentation fault happens.

Bibbidi-Babbidi-Boo, 2020-08-25:
1	../sysdeps/x86/dl-procinfo.c: No such file or directory.

Bibbidi-Babbidi-Boo, 2020-08-26:
@maximilianwulf Any idea what this could be? Is it possibly a version problems with the packages?

maximilianwulf, 2020-08-27:
Hey @Bibbidi-Babbidi-Boo hard to say at this point, can you share the core dump?

Bibbidi-Babbidi-Boo, 2020-09-01:
Attaching to process 9596
[New LWP 9607]
[New LWP 9608]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
0x00007f2f2b8aa13f in __GI___select (nfds=0, readfds=0x0, writefds=0x0, 
    exceptfds=0x0, timeout=0x7ffe1fc4ead0)
    at ../sysdeps/unix/sysv/linux/select.c:41
41	../sysdeps/unix/sysv/linux/select.c: No such file or directory.
(gdb) bt
#0  0x00007f2f2b8aa13f in __GI___select (nfds=0, readfds=0x0, writefds=0x0, 
    exceptfds=0x0, timeout=0x7ffe1fc4ead0)
    at ../sysdeps/unix/sysv/linux/select.c:41
#1  0x000056347d82463b in ?? ()
#2  0x000056347d78423a in PyEval_EvalFrameEx ()
#3  0x000056347d78186a in PyEval_EvalCodeEx ()
#4  0x000056347d789a3c in PyEval_EvalFrameEx ()
#5  0x000056347d78186a in PyEval_EvalCodeEx ()
#6  0x000056347d789a3c in PyEval_EvalFrameEx ()
#7  0x000056347d789016 in PyEval_EvalFrameEx ()
#8  0x000056347d78186a in PyEval_EvalCodeEx ()
#9  0x000056347d7893ee in PyEval_EvalFrameEx ()
#10 0x000056347d78186a in PyEval_EvalCodeEx ()
#11 0x000056347d7813a9 in PyEval_EvalCode ()
#12 0x000056347d7b219f in ?? ()
#13 0x000056347d7ad232 in PyRun_FileExFlags ()
#14 0x000056347d7acc5d in PyRun_SimpleFileExFlags ()
#15 0x000056347d75b706 in Py_Main ()
#16 0x00007f2f2b7b4b97 in __libc_start_main (main=0x56347d75b0e0 <main>, 
    argc=4, argv=0x7ffe1fc4f658, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7ffe1fc4f648)
    at ../csu/libc-start.c:310
#17 0x000056347d75affa in _start ()

maximilianwulf, 2020-09-03:
Hey, the pasted output does not really help, sorry. 

Which kernel version, OS version, ROS version, elevation mapping version and grid map version do you have?

Bibbidi-Babbidi-Boo, 2020-09-03:
Kernel version: Linux 5.4.0-42-generic x86_64
OS version: Ubuntu 18.04.5
ROS version: melodic
elevation mapping version: 0.7.0
grid_map version: 1.6.2