ANYbotics/elevation_mapping #140: catkin_make problem: argument discards qualifiers [-fpermissive]

Limiao813, 2020-10-21:
Hello, there is a problem when I catkin_make this repo.

I have build dependencies including kindr and grid_map successfully, and git clone kindr and elevation mapping in same catkin_ws/src.

when I catkin_make, it failed.
`/home/chaos/catkin_test_ws/src/elevation_mapping/elevation_mapping/src/input_sources/Input.cpp:36:32: error: passing ‘const XmlRpc::XmlRpcValue’ as ‘this’ argument discards qualifiers [-fpermissive]
     if (parameters[member.first].getType() != member.second) {
                                ^
In file included from /opt/ros/kinetic/include/xmlrpcpp/XmlRpc.h:34:0,
                 from /home/chaos/catkin_test_ws/src/elevation_mapping/elevation_mapping/include/elevation_mapping/input_sources/Input.hpp:11,
                 from /home/chaos/catkin_test_ws/src/elevation_mapping/elevation_mapping/src/input_sources/Input.cpp:9:
/opt/ros/kinetic/include/xmlrpcpp/XmlRpcValue.h:100:18: note:   in call to ‘XmlRpc::XmlRpcValue& XmlRpc::XmlRpcValue::operator[](const string&)’
     XmlRpcValue& operator[](std::string const& k) { assertStruct(); return (*_value.asStruct)[k]; }
                  ^
/home/chaos/catkin_test_ws/src/elevation_mapping/elevation_mapping/src/input_sources/Input.cpp:45:46: error: ambiguous overload for ‘operator[]’ (operand types are ‘const XmlRpc::XmlRpcValue’ and ‘const char [5]’)
   name_ = static_cast<std::string>(parameters["name"]);
`
Any comments would be helpful, thanks~~

Limiao813, 2020-10-21:
This is the code's problem? but I didn't change any code.

Limiao813, 2020-10-21:
okkkkk... 

This problem solved, I use kinetic and according to the error messages, I find `/opt/ros/kinetic/include/xmlrpcpp/XmlRpcValue.h` operator[ ] isn't const, however in `Input.cpp` the fuction's parameter is const. So this error message appeared.

I removed the `const` in the corresponding function, then compiled

this is not the best solution, if you have any good solutions or questions, please @ me

maximilianwulf, 2020-10-21:
Hello @Limiao813, we currently only support noetic, can you upgrade your ROS version?

Limiao813, 2020-10-21:
Hello @maximilianwulf,  My ROS version is kinetic, this error message didn't appear after remove `const`. Maybe I should upgrade...Thank you very much~~

closed 2020-10-22 by Limiao813