ANYbotics/elevation_mapping #141: rviz dies in simple_demo.launch

saiprakash-c, 2020-10-27:
**Configuration:** 

Ubuntu 18.04 
ROS-Melodic 

**Issue:** 

I ran `roslaunch elevation_mapping_demos simple_demo.launch`. The rviz node dies with the following message

`[rviz-4] process has died [pid 31135, exit code -11, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/vulcan/catkin_ws/src/elevation_mapping/elevation_mapping_demos/rviz/elevation_map_visualization.rviz __name:=rviz __log:=/home/vulcan/.ros/log/f4ac41e6-17eb-11eb-84e6-9848274778d7/rviz-4.log].
log file: /home/vulcan/.ros/log/f4ac41e6-17eb-11eb-84e6-9848274778d7/rviz-4*.log
`

I debugged using gdb. There appears to be a problem when `queue_size_property_->getInt()` is called in `grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h`. 

Here is the backtrace

[image]

I am unable to debug any further. Does anyone know what's going on here?

Thanks a lot!

maximilianwulf, 2020-10-28:
Hi @saiprakash-c, I don't get this issue with `noetic`, we currently support only noetic. Is it possible for you to switch? 

If not, could you share your setup in a docker container such that we could have a look?

maximilianwulf, 2020-10-28:
And the issue is the same as this one: https://github.com/ANYbotics/grid_map/issues/267

Will close because of duplicate.

saiprakash-c, 2020-10-29:
Thanks for the response, @maximilianwulf ! 

I will consider shifting to `noetic`. I never used Docker, so I will need some time to share my setup.

Austin-Hu, 2020-11-09:
Hi @saiprakash-c, @maximilianwulf and @StefanFabian,

I also reproduced the same issue (including `grid_map` Issue #[267](https://github.com/ANYbotics/grid_map/issues/267)), with exactly the same backtrace @saiprakash-c showed [here](https://github.com/ANYbotics/elevation_mapping/issues/141#issue-730061822), when clicking some `GridMap` topic (published by some ROS package we're developing) in rviz.

As @saiprakash-c and @StefanFabian analyzed, [queue_size_property_](https://github.com/ANYbotics/grid_map/blob/master/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h#L105) was NULL, although I confirmed that the [MessageFilterDisplay::MessageFilterDisplay()](https://github.com/ANYbotics/grid_map/blob/master/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h#L94) constructor was called before. So it's weird that its abstract base class [_RosTopicDisplay::_RosTopicDisplay()](https://github.com/ANYbotics/grid_map/blob/master/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h#L55) constructor (without argument) was never called?

BTW, in this case, the issue could be reproduced in either `melodic` or `neotic`.

Thanks!

maximilianwulf, 2020-11-11:
@Austin-Hu thanks for the input. 

Could you check if this line is executed? https://github.com/ANYbotics/grid_map/blob/master/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h#L60

Austin-Hu, 2020-11-11:
> @Austin-Hu thanks for the input.
> 
> Could you check if this line is executed? https://github.com/ANYbotics/grid_map/blob/master/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h#L60

No, the [_RosTopicDisplay::_RosTopicDisplay()](https://github.com/ANYbotics/grid_map/blob/master/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h#L55) constructor was never called, no matter I printed log there, or wanted to use gdb to step into.

maximilianwulf, 2020-11-11:
Ah okay, then it is clear. Could you check if you enter this constructor? https://github.com/ANYbotics/grid_map/blob/master/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h#L96

And if yes, add

```
    queue_size_property_ =
        new IntProperty("Queue Size", 10,
                        "Size of TF message filter queue.\n"
                        "Increasing this is useful if your TF data is delayed significantly "
                        "w.r.t. your data, but it can greatly increase memory usage as well.",
                        this, SLOT(updateQueueSize()));
    queue_size_property_->setMin(0);
```

maximilianwulf, 2020-11-11:
Or 

```
  MessageFilterDisplay() : tf_filter_(nullptr), messages_received_(0), _RosTopicDisplay()
  {
    QString message_type = QString::fromStdString(ros::message_traits::datatype<MessageType>());
    topic_property_->setMessageType(message_type);
    topic_property_->setDescription(message_type + " topic to subscribe to.");
  }
```

based on: https://stackoverflow.com/questions/120876/what-are-the-rules-for-calling-the-superclass-constructor

Austin-Hu, 2020-11-11:
@maximilianwulf, at first, I confirmed that the MessageFilterDisplay::MessageFilterDisplay() constructor was called, so I assume that its (abstract) base class constructor _RosTopicDisplay::_RosTopicDisplay() should be also called ahead, but it wasn't. So I also tried to explicitly call _RosTopicDisplay() in MessageFilterDisplay(), like what you suggested:
```
  MessageFilterDisplay() : tf_filter_(nullptr), messages_received_(0), _RosTopicDisplay()
```

But my added log in _RosTopicDisplay() couldn't be printed out, and `gdb` couldn't step into it, either.

maximilianwulf, 2020-11-11:
Hm, also it should just call the basic constructor of the superclass. 

This plugin is doing it the same way: https://github.com/ros-visualization/rviz/blob/8bd56fb928b255e095424a5f1fb3ff8a5274602e/src/rviz/default_plugin/effort_display.h

maximilianwulf, 2020-11-11:
Ah okay, but the original one does not initialize this member in the base class anymore: https://github.com/ros-visualization/rviz/blob/3cdc03a5186a53f644c5abd55b9971457bdffa1c/src/rviz/message_filter_display.h#L55

maximilianwulf, 2020-11-11:
@Austin-Hu can you try the first proposal from me?

Austin-Hu, 2020-11-11:
Sorry, currently I'm working on another issue (for our project), and I need to spend additional time to set up the reproducing environment.

But in theory, with your 1st proposal adopted, I'm afraid that another crashing would happen [here](https://github.com/ANYbotics/grid_map/blob/master/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h#L162), because the `unreliable_property_` object should have been created as well.

Austin-Hu, 2020-11-11:
BTW, if it's convinient for you, you can also reproduce the issue, by clicking any `GridMap` topic in rviz, if it can be list by any (melodic or neotic) ROS nodelet you're running.

maximilianwulf, 2020-11-11:
Okay, all clear. 

I couldn't reproduce it internally so far. But let's see.

StefanFabian, 2020-11-11:
Might I ask why you are using the rviz `MessageFilterDisplay`?
Why not create the copy in your own namespace which would avoid all issues caused by their implementation changing?

Austin-Hu, 2020-11-11:
@maximilianwulf, thanks! Once I can switch to the reproducing step, I can try your 1st proposal. But I think it'd be some experiment or work aroud. We need to check why _RosTopicDisplay::_RosTopicDisplay() couldn't be called as expected...

Austin-Hu, 2020-11-11:
@StefanFabian, I don't know. Maybe the [callstack](https://github.com/ANYbotics/elevation_mapping/issues/141#issue-730061822) from @saiprakash-c could answer your question.

StefanFabian, 2020-11-12:
@Austin-Hu I think you misunderstood what I was talking about.
The entire problem here is caused by this plugin using a copy of the MessageFilterDisplay from the rviz repository.
Since this copy has the same signature (class name and namespace), in the binary they refer to the same thing.
However, as it is loaded into the rviz binary, the class definition used is the one in the rviz repository.
Hence, if they change their definition, the two are not the same and the compiled plugin will try to use a declared interface that does not exist or as in this case member variables that do not exist anymore.
This is also the reason why you can't get gdb to stop in the constructor of `_RosTopicDisplay()` as this constructor is never called. I assume that the change in the original rviz `_RosTopicDisplay` causes the binary to symbol mapping to be invalid for this source file.

The obvious solution would be to change the namespace unless there is a reason why that isn't possible.
If it isn't possible, the alternative would be to update the copy to reflect the changes made in the rviz repository.
However, that will break again if they change it again and as it isn't part of the public API, they can freely do so.

mariethsch, 2020-11-25:
> Hi @saiprakash-c, I don't get this issue with `noetic`, we currently support only noetic. Is it possible for you to switch?
> 
> If not, could you share your setup in a docker container such that we could have a look?

I have the same issue. However I cannot switch to noetic ( I am also using Ros Melodic and Ubuntu 18.04) as other project parts cannot be transferred to noetic. Do you have a suggestion for me how I could try to implement it in melodic?

StefanFabian, 2020-11-25:
@mariethsch The quickest fix is to replace all calls to `queue_size_property_->getInt()` by a fixed constant, e.g., 10.
There may be other undefined behavior, though.
A better fix would be to change the namespace in that file from `rviz` to `elevation_mapping`.

mariethsch, 2020-11-25:
@StefanFabian Which way have you done it? 
I would like to fix it 'properly'. The question is now, when I have changed the namespace, I also have to change it in all subsequent files. So far I have identified GridMapDisplay.hpp, GridMapVisual.hpp who call the header file. But in there I have a namespaces called rviz again. So presume I would need to change these, too. 
Finally they are included in the GridMapDisplay.cpp where also original rviz- header files are included. So how do I know which of these classes/functions come from the original and which from the plugin?

StefanFabian, 2020-11-26:
I didn't have the time for a proper fix, so I went with the quick fix.
The proper solution would be not to use the namespace of a different project in your package.
I would advise against changing it in public headers as that might lead to incompatibilities with other packages but in the src folder there shouldn't be rviz namespaced definitions.
As to how to know which are copied from rviz and which are not, you'll have to look at the individual files and check if there is an equivalent file from rviz and how they differ (e.g. is it maybe just an old version).

mariethsch, 2020-11-27:
Thank you, I will see which solution is the best for our project.

maximilianwulf, 2020-11-30:
I would set the value manually to 1 and wait for the proper fix that will be submitted.

maximilianwulf, 2020-12-02:
I fix has been submitted to the grid_map rviz plugin. Please check if it solves this issue. 

Kudos to https://github.com/Magnusgaertner for submitting the fix.

maximilianwulf, 2020-12-21:
Feel free to re-open if there are further issues.

closed 2020-12-21 by maximilianwulf