ANYbotics/elevation_mapping #142: Pose Information never retrieved?

iamsavva, 2020-11-05:
ahoy! 

i'm running D435i on a TX2. I'm using VINS-Fusion-GPU as my VIO to get pose estimates: VIO pubilshes a nav_msg, I republish it as a stamped pose with cov, feed it into the elevation mapping, but for some reason it doesn't seem to be recognized. Could you help me with that, please?

Here are the errors and the debug messages I'm getting:

[image]

and here are the setup messages:
[image]

Here's the yaml that I'm running:
```
input_sources:
- name: front # A name to identify the input source
  type: pointcloud # Supported types: pointcloud
  topic: /camera/depth/color/points
  queue_size: 5
  publish_on_update: true # Wheter to publish the elevation map after a callback from this source. 

map_frame_id: body
robot_base_frame_id: body
track_point_frame_id: body
sensor_frame_id: camera_depth_optical_frame

robot_pose_topic: /util_node/util/pose
robot_pose_cache_size: 200
track_point_x: 0.0
track_point_y: 0.0
track_point_z: 0.0
time_tolerance: 1

robot_pose_cache_size: 10
```

Here's my tf_tree:

[image]

would appreciate any help!
Cheers,
Savva M

maximilianwulf, 2020-11-09:
Hey @iamsavva, you are missing the connection in the tf tree. `camera_link` should be attached to the `camera` node.

iamsavva, 2020-11-10:
@maximilianwulf, thanks for the note! I seem to be having the same error despite that. I am now using T265 with D435i.

I am having the same issue:
[image]

despite the seemingly correct tf_tree: 
[image]

it seems to be missing the pose topic. except that the topic is being published, here:
[image]

i noticed that the "subscribed to pointcloud topic" info message always appears, yet the "subscribed to pose topic" is not there. Is it supposed to be like this?
[image]

would appreciate any help. Many thanks in advance.

iamsavva, 2020-11-10:
here's my config file as well:
[image]

maximilianwulf, 2020-11-11:
At which frequency do you publish your pose? It could be that at specific timestamps there is no pose available.

Is there no elevation map visible at all?

maximilianwulf, 2020-12-21:
Feel free to re-open.

closed 2020-12-21 by maximilianwulf