ANYbotics/elevation_mapping #144: about quadruped demos

Allen-Debuger, 2020-12-01:
Thanks for your excellent package, I am woking on using this pkg on my quadruped robot but I have no idea till now. I wonder could you update some simple demos about quadruped robot such as your starleth or Anymal

maximilianwulf, 2020-12-01:
Hey @Allen-Debuger, did you try the turtlebot3 demo?

Allen-Debuger, 2020-12-01:
> Hey @Allen-Debuger, did you try the turtlebot3 demo?

yes, I have run your turtlebot demo, do you mean this package works in the same way on both quadruped robot and wheeled robot?

maximilianwulf, 2020-12-01:
>  do you mean this package works in the same way on both quadruped robot and wheeled robot?

Yes, you need to provide a pose of the robot, doesn't matter if quadruped or differential drive, and then you add a sensor to it.

Allen-Debuger, 2020-12-02:
> you need to provide a pose of the robot, doesn't matter if quadruped or differential drive, and then you add a sensor to it.

Honestly, I think wheeled and quadruped robot's locolization is quite different, so I wonder how to use kinematics and IMU to give the quadruped robot pose  if I want to use this pkg on my own robot. Hope more helpful information! Thanks!

maximilianwulf, 2020-12-02:
Hey @Allen-Debuger, yes I agree with you. This is not what I meant. The interface of elevation mapping requires a pose and a point cloud. 

This package does not make assumptions on the underlying state estimation method to come up with the pose. For different robot platforms, you will have to adapt the model for the estimation. Since this package is not about state estimation I cannot give further assistance in integrating a state estimation method. I can recommend https://github.com/ori-drs/pronto though, it should answer your question.

closed 2020-12-07 by maximilianwulf