ANYbotics/elevation_mapping #145: The error when build the elevation_mapping package

SUN-GE, 2020-12-08:
Thanks for your excellent work! 
I meet the error when I build the elevation_mapping package. I am using Ubuntu 18.04 with ROS melodic. And I have install the software which Elevation Mapping depends on: (Grid Map, kindr, kindr_ros, Point Cloud Library (PCL), Eigen. And catkin build them successfully.
But when I build the elevation map package. I meet the error below: 

Errors     << elevation_mapping:make /home/hebi/eleva_map_ws/logs/elevation_mapping/build.make.004.log
In file included from /home/hebi/eleva_map_ws/src/elevation_mapping/elevation_mapping/src/PointXYZRGBConfidenceRatio.cpp:12:0:
/home/hebi/eleva_map_ws/src/elevation_mapping/elevation_mapping/include/elevation_mapping/PointXYZRGBConfidenceRatio.hpp:25:5: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
     };
     ^
/home/hebi/eleva_map_ws/src/elevation_mapping/elevation_mapping/include/elevation_mapping/PointXYZRGBConfidenceRatio.hpp:28:3: error: ‘PCL_MAKE_ALIGNED_OPERATOR_NEW’ does not name a type; did you mean ‘EIGEN_MAKE_ALIGNED_OPERATOR_NEW’?
   PCL_MAKE_ALIGNED_OPERATOR_NEW
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   EIGEN_MAKE_ALIGNED_OPERATOR_NEW
make[2]: *** [CMakeFiles/elevation_mapping_pcl_types.dir/src/PointXYZRGBConfidenceRatio.cpp.o] Error 1
make[1]: *** [CMakeFiles/elevation_mapping_pcl_types.dir/all] Error 2
make: *** [all] Error 2
cd /home/hebi/eleva_map_ws/build/elevation_mapping; catkin build --get-env elevation_mapping | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << elevation_mapping:make           [ Exited with code 2 ]          
Failed    <<< elevation_mapping                [ 5.0 seconds ]                 
[build] Summary: 7 of 8 packages succeeded.                                    
[build]   Ignored:   13 packages were skipped or are blacklisted.              
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 7.2 seconds total.     

Could you pls help me with this issue? Thanks a lot.

usamex, 2020-12-09:
I had been looking for this as well and I guess there's no option but to implement your own costmap plugin. You can find an excellent example in the PR (Pull Request Section) #84 . Hope this helps.

maximilianwulf, 2020-12-09:
Hi @SUN-GE, @usamex is right, at the moment the package does not support costmaps. The grid_map package provides functionalities to convert height maps to costmaps, see [here](https://github.com/ANYbotics/grid_map/tree/master/grid_map_costmap_2d). 

For the moment, you will have to implement this feature by yourself. We do not plan to integrate it.

SUN-GE, 2020-12-10:
Thanks for your advice! @usamex @maximilianwulf

closed 2020-12-10 by SUN-GE

hzyhuter, 2022-11-15:
@usamex hey Thank you for giving me this advice, can you teach me the exact use of this plugin? I am new in this area. Appreciate it!