ANYbotics/elevation_mapping #146: The data in  '/elevation_mapping/elevation_map_raw' topic.

Psyduck123, 2020-12-09:
Hi, I have a question about the data in '/elevation_mapping/elevation_map_raw' topic. I subscribed to this topic and plot it as a grey image by transforming it into a NumPy array. I found that the image I plot is different from what I saw in Rviz (the obstacle position is different, sometimes the obstacle is been cut off).  

I am wondering the data in '/elevation_mapping/elevation_map_raw' topic is expressed in which frame (base_link or map)?  
Why the array I plot from the topic is different from the grid_map in Rviz in realtime.

Looking forward to your reply! Thanks.

maximilianwulf, 2020-12-09:
Can add print screens? 

Out of my head try [this function](https://github.com/ANYbotics/grid_map/blob/d9ee02deb5af682339610ad2313c0bd9c052df75/grid_map_core/include/grid_map_core/GridMap.hpp#L477) before you convert it to an image.

Psyduck123, 2020-12-09:
I use python matplotlib to save the image and check (cannot print screens). 
I wonder if there any function in python to get the elevation map in array under the world frame?

maximilianwulf, 2020-12-09:
You can check the map_frame_id in your configuration. There you can also switch it to any frame you want. 

I hope that helps.

Psyduck123, 2020-12-10:
I directly subscribe to the '/elevation_mapping/elevation_map_raw' topic. There is no map_frame_id in this topic. So I save the data in this topic as an array and plot it as an image. What I got from the Rviz of this topic and what I plot as shwon below.
[image]
[image]

Why the array I got is different from the arrangement in Rviz（from the same topic) ?

maximilianwulf, 2020-12-10:
Hey thanks for the images, that confirms my suspicion. 

Please use `[convertToDefaultStartIndex()](https://github.com/ANYbotics/grid_map/blob/d9ee02deb5af682339610ad2313c0bd9c052df75/grid_map_core/include/grid_map_core/GridMap.hpp#L477)` before you convert it to an image.

maximilianwulf, 2020-12-21:
Feel free to re-open.

closed 2020-12-21 by maximilianwulf