ANYbotics/elevation_mapping #147: Throwing std::bad_alloc for aparently no reason

caiofis, 2020-12-11:
I am using the package on some bags recorded on my robot equipped with a velodyne puck and it is working like a charm.  
But sometimes it just breaks throwing a std::bad_alloc and I am not able to trace where it is coming from.
Using Valgrind the error didn't occurred and using gdb I got the following error:

[image]

Does anyone faced this issue before?

maximilianwulf, 2020-12-11:
Hi @caiofis, can you `bt` in the gdb to see at what line the the software crashed?

caiofis, 2020-12-11:
Sure!
It seems to be happening in the publishFusedMapCallback, but it is weird that the program run for a while before crashing.
[image]

maximilianwulf, 2020-12-11:
Can you `list` in frame 8?

caiofis, 2020-12-11:
Here it goes
[image]

caiofis, 2020-12-11:
I was able to stop the std::bad_alloc using the hack proposed in #42, but using a hard-coded maximum.
My point cloud is quite sparse, I am wondering if the lack of observations on some cells for some time would make the error ellipse grow, resulting on a massive number of points fused, leading to the bad allocation.

My hack bellow line 312 of [ElevationMap.cpp](https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/src/ElevationMap.cpp#L311):
```cpp
if (maxNumberOfCellsToFuse > 1000) {
    // Skip fusing.
      fusedMap_.at("elevation", *areaIterator) = rawMapCopy.at("elevation", *areaIterator);
      fusedMap_.at("lower_bound", *areaIterator) = rawMapCopy.at("elevation", *areaIterator) - 2.0 * sqrt(rawMapCopy.at("variance", *areaIterator));
      fusedMap_.at("upper_bound", *areaIterator) = rawMapCopy.at("elevation", *areaIterator) + 2.0 * sqrt(rawMapCopy.at("variance", *areaIterator));
      fusedMap_.at("color", *areaIterator) = rawMapCopy.at("color", *areaIterator);
      continue;
    }
```
Image from the sparse point cloud on Rviz:
[image]

maximilianwulf, 2020-12-21:
Thank you for this observation. We will follow up on this. 

But for now you are not blocked?

caiofis, 2020-12-27:
I am not blocked anymore, thank you so much for your help. Please send me a notice when you figure out a more elegant solution to this issue. :)

closed 2020-12-27 by caiofis

maximilianwulf, 2021-01-06:
We will do :)