ANYbotics/elevation_mapping #148: how the camera types influence elevation map

Allen-Debuger, 2020-12-15:
I have tried several different structured-light camera config file (realsense435 & kinect) on my A1 robot, but the elevation map always outputs NAN on every cell, However, when I test perfect type sensor in the same scene with same other configurations, the elevation can be built normally. Hope you could tell me why and how to solve it! Thanks!
realsense435:
[image]
perfect:
[image]

maximilianwulf, 2020-12-16:
Hey, there was a small bug which is fixed now. You should get the non-NaN values now. 

Let me know if you have further issues.

closed 2020-12-16 by maximilianwulf

caffreyu, 2020-12-18:
I had the same problem with RealSense D430/435 and Melodic. We had the same problem with the updated code: we got NaN values with the structured_light sensor type, and we had to switch to the perfect sensor type to avoid that issue. Also, we encountered a similar issue as #138 with holes in the map. Could you please help us double-check on that? Thank you!

maximilianwulf, 2021-01-06:
Can you check it again with the newest version on master? It should actually be solved. We can follow up if it still persists.