ANYbotics/elevation_mapping #149: Configuration file error

aled96, 2020-12-16:
Hi, 
I am trying to use this configuration file:

```
input_sources:
    name: front # A name to identify the input source
    type: pointcloud # Supported types: pointcloud
    topic: /d435i_camera/depth/color/points
    queue_size: 5
    publish_on_update: true # Wheter to publish the elevation map after a callback from this source.
    
map_frame_id: map
sensor_frame_id: d435i_camera_depth_optical_frame
robot_base_frame_id: pelvis
robot_pose_with_covariance_topic: /T265_camera/pose/sample
robot_pose_cache_size: 200
track_point_frame_id: world
track_point_x: 0.0
track_point_y: 0.0
track_point_z: 0.0
time_tolerance: 1
```

But when I launch the file, I obtained this error:
`Input source must be specified as map, but is XmlRpcType:4.`

What I have to do to solve it ?
Thanks!

maximilianwulf, 2020-12-16:
Hey @aled96, we changed the yaml structure a bit, instead write

```
input_sources:
  front: # A name to identify the input source
    type: pointcloud # Supported types: pointcloud
```

sorry for the inconvenience.

closed 2020-12-16 by maximilianwulf

aled96, 2020-12-16:
Thank you @maximilianwulf !