ANYbotics/elevation_mapping #150: Elevation mapping dying due to Eigen assertion row >= 0 && row < rows() && col >= 0 && col < cols()

blablebli-robots, 2020-12-22:
Hello,

Revisiting the elevation mapping again. I'm running a simulation in gazebo using a rotating lidar as a sensor. I am running into the following error:
`elevation_mapping: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:365: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<float, -1, -1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = float; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.
[elevation_mapping-1] process has died [pid 15186, exit code -6, cmd /opt/ros/melodic/lib/elevation_mapping/elevation_mapping __name:=elevation_mapping`

Any ideas as to why I'm getting this error? The node runs for a while, but eventuelly fails with this message.

Cheers!

blablebli-robots, 2020-12-23:
Found the solution: In ElevationMapping.cpp, line 424 uses iterator, which is modified in the for loop (in line 417) without checking for valid, adding

if (!visibilityCleanupMap_.isValid(*iterator)) continue;

before line 424 solves the issue.

Cheers!

maximilianwulf, 2021-01-07:
Hey @blablebli-robots thank you for the issue. 

The line you mention does not directly point to a for loop, see https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/src/ElevationMapping.cpp#L424. 

Maybe line number changed.

blablebli-robots, 2021-01-07:
Sorry about the confusion, I meant file ElevationMap.cpp instead.

maximilianwulf, 2021-01-07:
But this for loop already has the check? https://github.com/ANYbotics/elevation_mapping/blob/master/elevation_mapping/src/ElevationMap.cpp#L431

blablebli-robots, 2021-01-07:
Argh, wrong line numbers, look at 451-462, iterator gets modified but not checked for validity.

maximilianwulf, 2021-01-07:
True, thank you. I will probably add this check then.