ANYbotics/elevation_mapping #151: Build error with: PCL_MAKE_ALIGNED_OPERATOR_NEW

alessandromorra, 2021-01-04:
I have just tried your demo and got this error: 

```
12:01 $ catkin build elevation_mapping_demos
-----------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/alessandromorra/arc_ws
-----------------------------------------------------------------
Build Space:        [exists] /home/alessandromorra/arc_ws/build
Devel Space:        [exists] /home/alessandromorra/arc_ws/devel
Install Space:      [unused] /home/alessandromorra/arc_ws/install
Log Space:          [exists] /home/alessandromorra/arc_ws/logs
Source Space:       [exists] /home/alessandromorra/arc_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------------
[build] Found '122' packages in 0.0 seconds.                                                                
[build] Package table is up to date.                                                                        
Starting  >>> kindr                                                                                         
Finished  <<< kindr                                  [ 0.1 seconds ]                                        
Starting  >>> kindr_ros                                                                                     
Finished  <<< kindr_ros                              [ 0.1 seconds ]                                        
Starting  >>> elevation_mapping                                                                             
______________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << elevation_mapping:make /home/alessandromorra/arc_ws/logs/elevation_mapping/build.make.002.log                                                                                                                                                                                                                                                                   
In file included from /home/alessandromorra/arc_ws/src/elevation_mapping/elevation_mapping/src/PointXYZRGBConfidenceRatio.cpp:12:
/home/alessandromorra/arc_ws/src/elevation_mapping/elevation_mapping/include/elevation_mapping/PointXYZRGBConfidenceRatio.hpp:25:5: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
   25 |     };
      |     ^
/home/alessandromorra/arc_ws/src/elevation_mapping/elevation_mapping/include/elevation_mapping/PointXYZRGBConfidenceRatio.hpp:28:3: error: ‘PCL_MAKE_ALIGNED_OPERATOR_NEW’ does not name a type; did you mean ‘EIGEN_MAKE_ALIGNED_OPERATOR_NEW’?
   28 |   PCL_MAKE_ALIGNED_OPERATOR_NEW
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   EIGEN_MAKE_ALIGNED_OPERATOR_NEW
make[2]: *** [CMakeFiles/elevation_mapping_pcl_types.dir/src/PointXYZRGBConfidenceRatio.cpp.o] Error 1
make[1]: *** [CMakeFiles/elevation_mapping_pcl_types.dir/all] Error 2
make: *** [all] Error 2
cd /home/alessandromorra/arc_ws/build/elevation_mapping; catkin build --get-env elevation_mapping | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..............................................................................................................................................................................................................................................................................................................................................................................
Failed     << elevation_mapping:make                 [ Exited with code 2 ]                                                                                                                                                                                                                                                                                                   
Failed    <<< elevation_mapping                      [ 4.1 seconds ]                                                                                                                                                                                                                                                                                                          
Abandoned <<< elevation_mapping_demos                [ Unrelated job failed ]                                                                                                                                                                                                                                                                                                 
[build] Summary: 2 of 4 packages succeeded.                                                                                                                                                                                                                                                                                                                                   
[build]   Ignored:   118 packages were skipped or are blacklisted.                                                                                                                                                                                                                                                                                                            
[build]   Warnings:  None.                                                                                                                                                                                                                                                                                                                                                    
[build]   Abandoned: 1 packages were abandoned.                                                                                                                                                                                                                                                                                                                               
[build]   Failed:    1 packages failed.                                                                                                                                                                                                                                                                                                                                       
[build] Runtime: 4.8 seconds total.             
```

Is this an error of the package or something wrong with my installation?
I am using Ubuntu 18.04.

With this fix it seems to work fine: https://github.com/ANYbotics/elevation_mapping/compare/master...alessandromorra:master

maximilianwulf, 2021-01-06:
Hey @alessandromorra can you try the fix from @heuristicus? 

Can you also tell me your PCL version?

alessandromorra, 2021-01-06:
I am using 1.8.1.
Unfortunately #152 does not fix it for me, I did not change my PCL version. 
@heuristicus did you do something else as well?

heuristicus, 2021-01-06:
The fix in the PR will only work in PCL versions 1.11.1 and above since that is where this macro was introduced. For older versions I think you have to do what you did and change it back to use the Eigen operator.

In my case I checked out the `release` branch instead of master and that worked, but if you need the latest version then I think that's the only way to fix it if you're stuck with older PCL versions.

alessandromorra, 2021-01-07:
I see, thanks for the clarification.

maximilianwulf, 2021-01-07:
Hm okay, I will check if we can add a version dependent preprocessor macro.

alessandromorra, 2021-01-08:
Yes I can update my PCL version, so this closes the issue for me. Many thanks for the support!

closed 2021-01-08 by alessandromorra

digiwang, 2021-03-05:
I had the same exact problem as @alessandromorra, but wasn't able to fix it for a while. So, for anyone else that runs into this issue down the road, all I had to do was build the newest version of pcl_ros alongside this repository (I initially had v1.7.1 installed). You should not have to make the change to **PointXYZRGBConfidenceRatio.hpp**.

Clone the `https://github.com/ros-perception/perception_pcl` repo within your **catkin_ws/src** folder (the latest version was v1.7.2 at the time of this writing). Then retry the build.

ThibaultRouillard, 2021-04-28:
Checking out to `release` worked for me. Thanks

Sunls666, 2021-09-30:
> The fix in the PR will only work in PCL versions 1.11.1 and above since that is where this macro was introduced. For older versions I think you have to do what you did and change it back to use the [...]

hi，i want to know how to checked out the `release`，thank you！

Sunls666, 2021-09-30:
“Checking out to release”，how to do it ？

hooram, 2021-10-16:
> “Checking out to release”，how to do it ？

`git checkout release`

ArduinoHocam, 2021-12-14:
Release branch of the elevation_mapping repo solved the issue for me! But why there are differences between master and release? That is weird. Can you also add this issue on README maybe for those of us who using different ros distro can be aware of this and maybe header file issue? @maximilianwulf  Thanks.